Atmosphere-NX/Atmosphere #1653: fatal error after having upload something on the sd

meuaxelexa, 2021-10-02:
## Bug Report
### What's the issue you encountered?
[ a fatal error occured when running atmosphere
program id: xxxxxxxxxxxxxx (replaced)
error desc: std::abort() called (0xffe)
report saved to /atmosphere/fatal_errors/report_0000XXXXX.bin]

I've uploaded files in the sd and that's it... in other folders than atmosphere
No change related to Atmosphère itself
Fixing archive bit with the tool does not work

### How can the issue be reproduced?
@every start

### Crash Report
[https://drive.google.com/file/d/1bai9p9jUpJpXPcFX5ADxuBUVOlsLm2w9/view?usp=sharing](url)

### System Firmware Version
can't access 

### Environment?
bootloader is hekate
atmosphere official release (but I can't load atmosphere to get more information)
installation not done by me

### Additional context?
na

masagrator, 2021-10-02:
Why did you erases those numbers? You think it's some confidential info? 😄

meuaxelexa, 2021-10-02:
> Why did you erased those numbers? You think it's some confidential info? 😄

Hi, no, just didn't take the time to copy them: is it of any use? 
Is there any tool we can use to read / parse the crash error bin?

masagrator, 2021-10-02:
> Is there any tool we can use to read / parse the crash error bin?

https://github.com/cathery/AFE_Parser

meuaxelexa, 2021-10-02:
top, but run on windows...  I'm on macos : let's now find a vm with windows ....

meuaxelexa, 2021-10-02:
and then: the program id: ID: 010041544D530000

HamletDuFromage, 2021-10-02:
> top, but run on windows... I'm on macos : let's now find a vm with windows ....

You can just compile it for macOS I would assume, I did so for Linux without any issue. Just install gcc and type make in your terminal inside the repository

SciresM, 2021-10-02:
Fix your archive bits.

Also, just look at the reports in a hex editor, it's what I do.

SciresM, 2021-10-02:
>I'm on macos

Yeah, macOS sets archive bits in a way that the switch can't accept.

You shouldn't use macOS to interact with switch sd card, it'll fuck stuff up.

closed 2021-10-02 by SciresM