Atmosphere-NX/Atmosphere #1654: Panic occurred while running Atmosphere

nooksta, 2021-10-04:
### What's the issue you encountered?

After moving all my files out of my sd card (I had previously used atmosphere on exFAT), reformatting to FAT32, then moving back into the sd card, when I try to boot atmosphere it gives me the error:
Panic occurred while running Atmosphere
Title ID : 0100000000000002b
Error: std::abort (0xFFE)
Hekate works normally, etc. Tried mounting system partition and deleting 80000000000000d1, tried deleting the contents folder and replacing my atmosphere folder with the newest release even though I already had it. Same thing pops up.

### How can the issue be reproduced?

Not sure/possibly same thing I did.

### Crash Report

[](https://gist.github.com/nooksta/4c70d5faa360789042a1930e8b15c1d8)

### System Firmware Version

newest version

### Environment?

- hekate, official release
- homebrew software: tesla, sys-clk, sys-ftpd-light, hbmenu, hbappstore, breeze, KosmosToolbox, EdiZon-SE, Goldleaf, 
nxdumptool, NxThemes Installer, ThemezerNX, SwitchPresence-Rewritten, retroarch
- SysNAND

Aura314, 2021-10-04:
you are such a hero you are doing piracy on your main memory the switch you are banned piracy is not supported.

masagrator, 2021-10-04:
> you are such a hero you are doing piracy on your main memory the switch you are banned piracy is not supported.

Are you okay? Your attitude is not commendable.

nooksta, 2021-10-04:
> you are such a hero you are doing piracy on your main memory the switch you are banned piracy is not supported.

I haven't installed any nsps or done any piracy at all. I honestly don't know what some of my homebrew apps even do if some are for that purpose since I got a lot of them through sdsetup.

fincs, 2021-10-04:
Bundles redistributes by third parties are notorious for containing all sorts of junk (some of it piracy related) that can interfere and cause system stability problems. Please try again with a clean installation using the [official release binaries](https://github.com/Atmosphere-NX/Atmosphere/releases/latest), and reopen the issue if the problem persists.

closed 2021-10-04 by fincs