Atmosphere-NX/Atmosphere #1657: atmosphere/fusee-secondary.bin is missing from all releases since 0.20.1

holdingsllc, 2021-10-05:
## Bug Report

### What's the issue you encountered?

`atmosphere/fusee-secondary.bin` is missing from all releases since `0.20.1`. No changes were made by me.

### How can the issue be reproduced?

Download the ZIP files from the Releases page and look in each ZIP files' `atmosphere/` folder for `fusee-secondary.bin`.

### Crash Report

N/A

### System Firmware Version

N/A

### Environment?

N/A

### Additional context?

N/A

SciresM, 2021-10-05:
This isn't a bug.

Fusee was re-written, fusee-secondary is no longer used and no longer exists.

Please read the release notes for 1.0.0...

SciresM, 2021-10-05:
Please note that the archive file which `fusee.bin` (`fusee-primary` also no longer exists, there is only one fusee program now) is located at `atmosphere/package3`.

Zack6286, 2025-01-26:
when i inject the payload on my switch it says secondary bin file is missing is there any fixes to it

HunterMario, 2025-01-26:
This most likely suggests that there is an issue with the payload you are attempting to inject to launch Atmosphere.

This is not a support site, it is for bug reports and other issues related to Atmosphere. If you need support with Atmosphere, please ask on the ReSwitched Discord server: [https://discord.gg/ZdqEhed](https://discord.gg/ZdqEhed)