Atmosphere-NX/Atmosphere #1660: Black Screen after update to Atmospere 1.1

Archik1000, 2021-10-09:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

I am Using Atmosphere 0.19.5 with Firmware version 11.0.1. I  have tried to update the Atmosfere first to 1.1,. Copy released Atmosphere to the SD card and Switch folder including latest homebrew. Put the latest bootloader folder as well. Using ReSwitched to load Atmosphere. After show me the logo of atmosphere it just give me black screen. Put latest hekate 5.6.3 and result the same. Purt fusse on SD, same result. I attache picture with root of my SD card. Please tell me how to fix it. Maybe Atmosphere 1.1 does not working with Firmware 110.1. I afraid to update the Firmware firs, causerie
[image]
 then no way back.  </br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

11.0.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Archik1000, 2021-10-10:
DogyDHW. How fusee works? As I understood I must do something on side of the switch instead of joystick.

Masamune3210, 2021-10-10:
Not even slightly the right place to be asking for help, go do more research. Google is your friend