Atmosphere-NX/Atmosphere #1684: Fatal Error 1.2.2

admiralkikan, 2021-10-31:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

A fatal error occurred when running Atmosphére. Title ID: 010041544d530000 Error Desc: std::abort() called (0xffe)

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?
Boot into Hekate
boot atmosphere in EmuMMC  

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report
Sorry, github kept giving my zip an error.

https://mega.nz/file/WvxgURjb#cpTeUi1jTkguVJ8ZsBbCzF92AUxIF1Snkar8wn_RCTI

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version
12.1.0

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?
Hekate
Atmosphère 1.2.2
sigpatches from itotaljustice
EmuMMC is partitioned based

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

This was after having fixed sigpatch issue after updating Atmosphère from 1.0. Received an error in regards to 010041544f5300 even though the folder doesn't exist anywhere. I can only assume it's failing to create prodinfo backup. Card isn't corrupted, tried hekate bit fixer which resulted in no issues found. And I'm on windows so macOS solutions won't work. I did try a fresh Atmosphère folder, which landed me the same issue. The only thing I didn't try was restoring emuMMC. My prodinfo is encrypted though if that means anything. 

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-10-31:
Fix your archive bits, this is an archive bit problem.

closed 2021-10-31 by SciresM

admiralkikan, 2021-10-31:
Interesting because I wouldn't have opened a ticket if I hadn't already tried to fix my archive bits. Whether through Hekate or manually setting/unsetting the archive box in properties settings. But thanks for the help I guess?

takoyaki99, 2021-10-31:
You also didn’t fill out the help sheet correctly.

takoyaki99, 2021-10-31:
Actually never mind you have enough info, probably. But, if that didn’t help, try asking the switchroot discord?