Atmosphere-NX/Atmosphere #1685: Panic occurred while running Atmosphere. Title ID: 010041544D530000. Error: std::abort (0xFFE)

Zaeehh, 2021-10-31:
## Bug Report

### What's the issue you encountered?

When I try to open Smash Ultimate, it loads into an error screen that says 
Panic occurred while running Atmosphere. 
Title ID: 010041544D530000. 
Error: std::abort (0xFFE)

### How can the issue be reproduced?

1. Boot into Hekate
2. Launch
3. Sysnand
4. Open Smash Ultimate

### Crash Report

Report from fatal_errors folder
https://drive.google.com/file/d/18ZBIV9rpTWdWo2G_gR2heqH6YUNciGTa/view?usp=sharing

### System Firmware Version

Switch version 13.1.0
Atmosphere 1.2.2

### Additional context?

I believe that I was on Switch version 13.0 when Atmosphere was working properly. When 13.1.0 came out, I accidently allowed the Switch to update. After it updated, Atmosphere wouldn't work anymore. I tried to update Atmosphere and Hekate to the latest version and it still wouldn't work. I get an error screen each time I open Ultimate.
Hekate's Fix Archive Bit button doesn't do anything. My SD Card is Fat32

takoyaki99, 2021-10-31:
I think this is an issue with your Skyline/Training Modpack, lots of other people have had this too.

Zaeehh, 2021-10-31:
Yep, that fixed the issue for me. Thank you so much!

closed 2021-10-31 by Zaeehh