Atmosphere-NX/Atmosphere #1686: Panic occurred while running Atmosphere. [Title ID: 0100000000000002B] [Error: Std::abort <0xFFE>]

lmysid, 2021-10-31:
## Bug Report

### What's the issue you encountered?

[I try to update the CFW as well as the atmosphere to the latest version. Can successfully launch Hekate but cannot boot into EmuMMC.]</br>
[I download the atmosphere 1.2.2, hekate v5.6.5]</br>

### How can the issue be reproduced?

[ * ]</br>
[1. Use TegraRcmGUI to get into Hekate]</br>
[2. Click Launch]</br>
[3. Click Atmosphere FSSO EmuMMC]</br>

### Crash Report

[report_0998a094.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7448990/report_0998a094.zip)

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 
  - [ Official release version 5.6.5 ]

SciresM, 2021-11-01:
You are not using atmosphere-1.2.2. You are using a super ancient version of atmosphere.

Make sure you delete whatever's inside your atmosphere/contents dir, you probably have old legacy atmosphere contents from pre-0.19 overriding latest.

closed 2021-11-01 by SciresM

lmysid, 2021-11-01:
Thanks! It works well now!
Previously is a friend helping me update but he cannot help me for now and I didn't update (and not even know how to) it for long until now. Not even know some old files may lead to this issue...
Once again, thanks!

SciresM, 2021-11-01:
No problem. Glad it's working now!