Atmosphere-NX/Atmosphere #1687: Fatal error when trying to run smash with mods

Xehanort88, 2021-10-31:
Crashes to atmosphere error 010041544d530000, Error: std::abort (0xFFE)

This when trying to start Smash with mods, installed with Quasar, which don't work either.
Attached Fatal error crashes.

[fatal error.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7449100/fatal.error.zip)
]

masagrator, 2021-10-31:
1. You didn't filled issue template
2. If you are using Skyline, update it to newest version from Actions tab on their github repo page

Xehanort88, 2021-10-31:
I didn't know at all how to fix it, as that is literally all the information I had.
I am using skyline, that is correct.

takoyaki99, 2021-10-31:
Update Atmosphere.
Update Skyline.
Update ARCropolis.
Update your Ultimate Training Modpack.
If you do all those it’ll be fixed.

Xehanort88, 2021-10-31:
I don't have the ultimate training modpack but I did update the other 3.

Sent from my Samsung Galaxy smartphone and typed with big fingers on a
small touchscreen. Please excuse any typographical errors.

closed 2021-11-01 by SciresM