Atmosphere-NX/Atmosphere #1691: Atmosphere Crashes when i try to boot up Smash Ultimate with ARCropolis

2ndstatus, 2021-11-01:
## Bug Report

### What's the issue you encountered?
I've been trying to launch Smash Ultimate Cart with Arcropolis and I keep getting this error:</br>

"A fatal error occurred when running Atmosphere.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)"</br>

The game boots normally if I remove Arcropolis from Atmospheree > Contents.

### How can the issue be reproduced?

Boot into RCM with Hekate Payload > Launch Atmosphere SysNand > Start Ultimate with ARCropolis inside SD Card. > Error.</br>

### Crash Report
https://gist.github.com/2ndstatus/43cb0e20a03fc7a6592d9dfe26ce815f</br>
### System Firmware Version

Firmware 13.1.0</br>
### Environment?

- Atmosphere via Hekate v5.6.5 Nyx 1.1.1
- Official release or unofficial build: Official
- EmuMMC or SysNAND: Affects both Emu & Sys (Partition Based)

### Additional context?

- Additional info about your environment: SD Card has 2 Partitions. First Partition is Fat32 Primary, Second Partition is ExFat Logical (Switch Automatically makes the 2nd Patition ExFat). I've already done a complete wipe and reinstall of all components, a few other programs like Checkpoint don't work either. I've tried Arc Bit Fix from Hekate, did not solve the issue.

SciresM, 2021-11-01:
update skyline.

closed 2021-11-01 by SciresM

2ndstatus, 2021-11-01:
> update skyline.

Thank you for the quick response, what is Skyline? Is there a Link available for the download?

takoyaki99, 2021-11-01:
Look on their github page—it’s a plugin thing for Acropolis, which was included with arcropolis. Now it isn’t bundled (I think) so you need a manual update

2ndstatus, 2021-11-01:
> Look on their github page—it’s a plugin thing for Acropolis, which was included with arcropolis. Now it isn’t bundled (I think) so you need a manual update
> […](#)

Alright- I read through the Arc Github and saw the notes about Skyline. I downloaded the Beta DL and... IT WORKED! It looks like the loading is really long but it worked! Thank you guys for the help.