Atmosphere-NX/Atmosphere #1692: When I am using emuMMC, the Smash Ultimate character selecting screen(CSS) go very slow

ztf147, 2021-11-02:
## Bug Report

### What's the issue you encountered?
When I am using emuMMC, the Smash Ultimate character selecting screen(known as CSS) go very slow.

### How can the issue be reproduced?
Yes, just use emuMMC and open Smash Ultimate to CSS you can see it happen.

it will happen even you have only 8 characters to choose.
It will happen no matter  you have mod or not.
it will happen no matter you overclock or not. 
It It won't happen in the real system with atmosphere even using mod by using the same SD card with the same soft environment.

### System Firmware Version
NS 13.0 and earlier.
atmosphere 1.1.1 and earlier

### Environment?
not using the Official release, but as I said, it won't happen in real system with atmosphere in the very same environment. 

### Additional context?

This bug may seems no big deal, but you know how often people open this game worldwide, as myself is a mod maker, so my job is to test my mod time and time again and it just killing me every time i test my mod on a switch. so i just really hope this can be solved ,thanks.

masagrator, 2021-11-02:
Update to newest Atmosphère before writing issue.
Noone cares about older versions as this bug may be already fixed.

ztf147, 2021-11-02:
> Update to newest Atmosphère before writing issue.
> Noone cares about older versions as this bug may be already fixed.

Tested in the 1.2.2 now, and changed the name of 0100A800016E000 to avoid the affect of other things, and problem remains. 
btw, also installed game to the NS itself not on the TF card, and problem remains.

SciresM, 2021-11-02:
This is super out of scope for me. Make an issue on the emummc repo.

closed 2021-11-02 by SciresM