Atmosphere-NX/Atmosphere #1694: After updating to last version games stop showing

jcenturion18, 2021-11-02:
Hi to everyone, I don't know if this is an actual problem of `Atmosphere` but I'm running out of ideas.
I was in Atmosphere 0.19.1 (I think) with OS 12.0.1, using fusee primary and `hekate`. Update to last version of everything in order to update the console too. 

After updating Atmosphere and Hekate, the games that I had in the emummc stop showing and only shows the system ones. I didn't touch any of the nintendo folders

Masamune3210, 2021-11-02:
Please go back and actually use the issue template, its there for a reason

jcenturion18, 2021-11-02:
> Please go back and actually use the issue template, its there for a reason

@Masamune3210 I'm really sorry about that. I promise I didn't delete it. It didn't show up. I'll close this one and open a new one with the template.

Again, sorry

closed 2021-11-02 by jcenturion18