Atmosphere-NX/Atmosphere #1696: Crash when trying to load mods on Puyo Puyo Tetris 2

pepsypollo, 2021-11-04:
## Bug Report

### What's the issue you encountered?

A fatal error occurred when running Atmosphére
Program ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

Report saved to /atmosphere/fatal_errors/report_000000013113ff6b.bin
Press POWER to reboot.

### How can the issue be reproduced?

Trying to load anything via LayeredFS on Puyo Puyo Tetris 2 even if its only an empty folder
(ex. 010038E011940000/romfs/raw)

### Crash Report
[report.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7475834/report.zip)

### System Firmware Version

13.1.0</br>

### Environment

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: 1.2.3
- Do you have additional kips or sysmodules you're loading: sys-ftpd-light MissionControl ovlloader
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: SysNAND

### Additional context?

- Additional info about your environment:
- Tried with game installed on nand and sd

Worked fine on 13.0.0 and atmosphere 1.1.1

SciresM, 2021-11-04:
Please post a screenshot of your atmosphere/contents folder.

ResultTargetLocked indicates concurrent SD card accesses are taking place.

pepsypollo, 2021-11-04:
[image]
The --- folders are empty just to track game ids I tried with and without them and it always crashes

Here's a report wihout them i dont know if they are the same btw how do you read them?
[new_report.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7476159/new_report.zip)

SciresM, 2021-11-04:
Does the error occur without Mission Control/ovlloader/sys-ftpd?

pepsypollo, 2021-11-04:
Yep
[report.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7476218/report.zip)

SciresM, 2021-11-04:
Do you have any kind of logging turned on in ams?

Are any erpt_reports being created?

SciresM, 2021-11-04:
@pepsypollo Please join the ReSwitched discord (or DM me, `SciresM#0524`) if you're interested in testing potential resolutions to this.

SciresM, 2021-11-04:
Resolved in ams 1.2.4.

closed 2021-11-04 by SciresM