Atmosphere-NX/Atmosphere #1697: Animal Crossing New Horizon version 2, crash after selecting 2nd player

programatix, 2021-11-05:
## Bug Report

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
Start Animal Crossing version 2. Game runs ok. On inviting 2nd player, system crash.

[ Did you make any changes related to Atmosphère itself? ]</br>
No.

[ If so, make sure to include details relating to what exactly you changed. ]</br>
N/A

### How can the issue be reproduced?
1. Run Animal Crossing New Horizon
2. In the game, invite 2nd player
3. Game crash right after screen transmission to black on selecting 2nd player

### Crash Report
https://gist.github.com/programatix/b6c97f74ee563b09d9546087501b27a4

### System Firmware Version
13.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
hekate_ctcaer_5.6.5.bin + fusee.

- Official release or unofficial build: 
Official release version 1.2.4

- Do you have additional kips or sysmodules you're loading:
No

- Homebrew software installed: [ * ]
N/A

- EmuMMC or SysNAND:
SysNAND

### Additional context?
N/A

iUltimateLP, 2021-11-05:
With second player you mean playing with another console over LAN or a local player on your console?

programatix, 2021-11-05:
Hi,

Local player with 2nd controller.

SciresM, 2021-11-05:
I am, uh, pretty confident that nothing Atmosphere does can cause es::ResultInvalidParameter() inside es, and I'm not sure what's going on with audio getting i2c::ResultNoAck() either.

Does the error occur playing on "stock" (non-atmosphere)?

programatix, 2021-11-05:
The crash screen always show error code 2145-0011. However, when I checked the crash folder, it would always followed by a few more error log files. They are mostly different errors.

I haven't test on "stock".

SciresM, 2021-11-05:
Please test stock. What result do you get?

SciresM, 2021-11-05:
Okay, I looked into this.

Here's my best understanding of what's happening:

* nn::ndrm::low::detail::INdrmLowUserInterface::Command2() is called, this eventually calls nn::es::IActiveRightsContext->Command14().
* nn::es::IActiveRightsContext() cmd 14 returns error InvalidParameter
* es throws fatal
* fatal sysmodule shuts down the i2c audio device
* audio sysmodule throws fatal due to no ack from shutdown i2c audio device
* NS gets "interface closed" error due to es or audio being terminated, throws fatal.

This explains all your error reports.

However, "IActiveRightsContext returns InvalidParameter"...no offense, but this seems like piracy to me, and piracy is not supported.

Going to tentatively close this issue and file it under "yeah this doesn't work when you're playing with pirated games", which like cool and okay go buy the game on the eShop.

If somebody reproduces this with a verifiably purchased-from-eShop copy of the game, please let me know.

SciresM, 2021-11-05:
(If you trigger this with an eShop copy of the game, DM `SciresM#0524` on discord and I'll look more into it.)

closed 2021-11-05 by SciresM