Atmosphere-NX/Atmosphere #1698: the console get stuck after switch logo

Kronos2308, 2021-11-05:
## Bug Report

ussing AMS 1.2.4 if i aplly a custom template for the home menu
 the console get stuck after switch logo
 and require to force the power off</br>

* using ams 1.2.4 and aplly a custom theme form home
* only happend in 1.2.4 
* * if i copy 1.2.3 or 1.2.2 back the console boot has normal

i am using AMS from the release in FW 13.1.0
and i with extra serices enabled and with pure AMS, same result te console hang after switch logo

### BTW
```
my user
Kronos2308#7751 , 493178350123089930
get banned. And after reading the rules again i see was unfair
i was asking about how make work a homebrew in nsp format (to get full ram), 
not about piracy. 
can be appealed the decision?

```

SciresM, 2021-11-05:
1) You didn't use the issue templates

2) Delete your custom sysmodules

3) no

closed 2021-11-05 by SciresM

Kronos2308, 2021-11-05:
[image]

Masamune3210, 2021-11-06:
"I didn't read the rules" isn't an excuse for breaking them

SciresM, 2021-11-06:
Please do not antagonize people in resolved issues.

Masamune3210, 2021-11-06:
Sorry

Kronos2308, 2021-11-06:
Any way
Has i say just using a layered fa no Home menu title make the console hang after switch logo so is maybe something with the changes on 1.2.4

SciresM, 2021-11-06:
This is, if anything, an issue with your theme/whatever.

Kronos2308, 2021-11-06:
No, it happens with any theme

SciresM, 2021-11-06:
Sure, sounds like an issue with themes. Probably not an atmosphere issue.

Kronos2308, 2021-11-06:
ok 
srry then

Kronos2308, 2021-11-06:
[image]

btc08gh, 2021-11-15:
Had the same issue with NXThemes on 1.2.4 (1.2.2 works perfectly btw.) Turned it to be another homebrew I had installed.

For some reason I can run both homebrews at the same time on 1.2.2 but not on 1.2.4.

Try moving everything out of /atmosphere/contents/ and putting each directory back in a handful at a time to figure out which one is stopping you from using 1.2.4 with NXThemes.

I finally narrowed it down to just these two homebrews on a clean install.  Once I removed the conflicting service ot of my contents folder the latest NXThemes works perfectly with 1.2.4.

btc08gh, 2021-11-15:
# Changelog
 ## 1.2.4
 + Changes were made to the way fs.mitm builds images when providing a layeredfs romfs.
   + Cache management (to avoid unnecessary rebuild) was revised, to add a grace period of ~500ms-1s between process closing romfs image and ams.mitm needing to rebuild if romfs is re-opened.
     + This makes our cache much more effective, previously we were re-building romfs several times.
   + RomFS image ownership was overhauled, with a new reference-counting implementation added (used to implement the above grace period).
   

I’m thinking this might be related since they are both using fs.mitm.

btc08gh, 2021-11-15:
What sys modules are you running besides NXThemes?  (Not interested in hearing about any piracy…just homebrew.)

btc08gh, 2021-11-15:
Confirmed the sys module clash does not happen when upgrading my 1.2.2 to 1.2.3.

NXThemes and SoundNX both run simultaneously without issues.

On 1.2.4 either one will run perfectly by itself, but not at the same time.

I can reliably reproduce in a clean environment.

Want me to fill out a separate issue or tell the sys module authors they need to accommodate the change?

Kronos2308, 2021-11-15:
layered fs on 
0100000000001000
010000000000100C
hang the overlay of the console on 1.2.4

[contents_old.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7539992/contents_old.zip)

btc08gh, 2021-11-15:
Exactly the same as my issue man.  100%

But 010000000000100C is just using fs.mitm to replace 3 sound files.

Something sus going on here.