Atmosphere-NX/Atmosphere #1699: "Panic occurred while running Atmosphere" when launching into emummc.

DinnerBug, 2021-11-05:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

I just tried updating Atmosphere, Hekate and the firmware using [this](https://nh-server.github.io/switch-guide/extras/updating/) guide. I downloaded the latest versions of both Hekate and Atmosphere and copied the files to the root of my sd card which is formatted to Fat32. I then used the latest version of TegraExplorer to dump the firmware. At last I tried launching Atmosphere through Hekate but encountered the error:

Panic occurred while running atmosphere.

Title ID: 0100000000000008
Error: std : : abort (0xFFE)

fss0 wrong path or Mismatched atmo files

Report saved to /atmosphere/fatal_errors/report_06861cef.bin

Press POWER to continue.

### Crash Report

Not a crash report, a fatal error report. I'm sorry, I couldn't figure out how to upload the report to the gist.
https://drive.google.com/file/d/19kwgw1MVPp4O0t8nMxjh1JsYuxiXc0W5/view?usp=sharing

### System Firmware Version

13.1.0 on stock
12.x on emummc (I currently cannot boot into athmosphere and don't remember which firmware version I was on)

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: Official release of Atmosphère
- Do you have additional kips or sysmodules you're loading: yes, I've tried launching athmosphere both with and wothout them but couldn't solve the issue.
- Homebrew software installed: awoo-installer, breeze, gamecard_installer, Goldleaf, NX-Activity-log, NXGallery, NX-Shell, Turnips, JKSV, ftpd, N-xplorer, NXThemeInstaller (I currently have a theme installed), SwitchPresence, sys-clk, tesla overlay, emuibo, Status-Monitor-Overlay, Mission control.
- EmuMMC or SysNAND: EmuMMC, partition-based

### Additional context?

How are you?

SciresM, 2021-11-05:
You have really outdated atmosphere. Delete your atmosphere dir, freshly re-extract latest release zip.

closed 2021-11-05 by SciresM

DinnerBug, 2021-11-05:
> You have really outdated atmosphere. Delete your atmosphere dir, freshly re-extract latest release zip.

Thank you so much for the quick response. I hope you have a nice day.