Atmosphere-NX/Atmosphere #1701: emuMMC no longer allows me to boot stock

Hwilliams8548, 2021-11-06:
## Bug Report

### What's the issue you encountered?
I formatted my SD card and then loaded up hekate to make an emuMMC, I made my emuMMC and booted it. I wanted to make sure I was in emuMMC so I deleted a title and then rebooted my switch only to find I was at a black screen (no not rcm, just a black screen) I thought it was since I just exited the cfw so I hard shut down (power for 12 secs) and it still black screened. I can boot CFW on my emuMMC and sysMMC but not stock.

### How can the issue be reproduced?
1. Open hekate
2. make an emuMMC
3. Not be able to boot stock

### System Firmware Version

12.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 1.2.4
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Daybreak and reboot to payload
- EmuMMC or SysNAND: file based

SciresM, 2021-11-06:
"stock" 

Why are you making an issue on the atmosphere repository about stock, when atmosphere is necessarily not stock?

You could make an issue on the emummc or hekate repositories, I guess, but fusee/atmosphere don't support booting stock at all.

closed 2021-11-06 by SciresM

Hwilliams8548, 2021-11-06:
No, I cant boot my switch without a jig. That's what I meant.

SciresM, 2021-11-06:
That doesn't make sense given your issue description.

"Stock" would be sysmmc, not emummc.

And "can't boot without a jig"...this is normal? Uninstall autorcm if you want that.

Hwilliams8548, 2021-11-06:
I dont have auto rcm, I dont even go into RCM I just get a black screen, My injector does not inject unless I do A force power off and then manually go to rcm. And the emuMMC part is because this did not occur until a made an emuMMC. Did you even read it?

SciresM, 2021-11-06:
I did read it. I am completely convinced this is user issues or at a minimum not Atmosphere.

This issue tracker is for bugs, not for user support. Try the Nintendo-Homebrew discord for user support.