Atmosphere-NX/Atmosphere #1702: Error Code: 2002-3502 (0x1b5c02) Program 0100000000001000

LukeC17, 2021-11-07:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ My friend powered off my console entirely (hold power, select power off) ]</br>
[ No, emmunand I'm pretty sure my emmunand is 12.0.3 and so is sysnand ]</br>
[ N/A]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Boot into RCM, inject fusee primary. I tried booting with hekate aswell to get into atmosphere (1.2.4 btw) and same thing. It boots all the way through the atmosphere logos to the switch logo which goes away, and just as you would think the home screen would appear, it crashes.]</br>

### Crash Report

[ N/A ]</br>
[N/A ]</br>
[ N/A]</br>

### System Firmware Version

X.X.X</br>
[ 12.0.3 SysNand, 12.0.3 EmuNand ]</br>
[ N/A ]</br>
[ N/A]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [I tried TegraExplorer and it seems to be able to open EMMC and EMUMMC folders. I feel like it is corrupted still somehow, but I don't know how to figure that out. I would at least like to get my saves back.]

LukeC17, 2021-11-07:
[image]

masagrator, 2021-11-07:
This is qlaunch titleid. You don't have any theme installed or qlaunch replacement like uLaunch?

SciresM, 2021-11-07:
2002-3502 is fs::ResultPortMmcNotActivated().

Sounds to me more like this guy's emummc is fucked, than anything. 

Sounds like SD corruption to me, I'm going to close this as not an atmosphere issue, I recommend re-formatting your SD as FAT32 and re-creating your emummc.

closed 2021-11-07 by SciresM

LukeC17, 2021-11-07:
> 2002-3502 is fs::ResultPortMmcNotActivated().
> 
> Sounds to me more like this guy's emummc is fucked, than anything.
> 
> Sounds like SD corruption to me, I'm going to close this as not an atmosphere issue, I recommend re-formatting your SD as FAT32 and re-creating your emummc.

My SD was FAT32, but yeah, I've been reading that re-creating emummc is what I have to do. Is there any way to get my saves back before I do this?

Masamune3210, 2021-11-07:
Can extract them manually on a PC but that's out of scope for this issue