Atmosphere-NX/Atmosphere #1705: Undock Controller to Connecting

DarkRaGaming, 2021-11-08:
What's the issue you encountered?
Hello, I am having an issue with my joy-cons not staying connected to my switch when undocking them from handheld mode. , I've tried using 2 different pairs of switch Joy-Cons and 1 Switch Controller, which were working fine before, This problem seems to happen with every other update for me. When in OFW it works fine. 

added:
Sys-Bot latest
Checkpoint
EdiZon SE
Goldleaf
Hb App Store
JKSV
Nxdumptool
Sys clk Management

How can the issue be reproduced?
Undocking switch controller from the switch

Crash Report
N/A My switch hasn't crashed.

System Firmware Version
13.1.0

Environment?

Atmosphere v1.2.4

The bootloader I use is Hekate v5.6.5

Software added:
Sys-Bot latest
Checkpoint
EdiZon SE
Goldleaf
Hb App Store
JKSV
Nxdumptool
Sys clk Management

Additional context?
If you need any more info, just ask. I have also Kept the same software on each update. It has happened to be previous in other versions but it normally gets fixed in the next update.

SciresM, 2021-11-08:
1. Does the issue occur without custom sysmodules?

2. Does the issue occur on stock without atmosphere?

Also not sure if this is a duplicate of #1704.

DarkRaGaming, 2021-11-09:
> 1. Does the issue occur without custom sysmodules?
> 
> 2. Does the issue occur on stock without atmosphere?
> 
> Also not sure if this is a duplicate of #1704.

I did put it in their at top that OFW (stock )  and before doing cfw joy cons worked. And this is not duplicate of #1704 . I did see their but it a different matter I believe as their is joy con's not connecting or disconnect .

Ch0cola, 2021-11-14:
Can confirm this bug, if you have paired the joycons on OFW and you switch to CFW, when you put out the joycons have problems to connect to Switch through BT.

To fix this, you should enter on system settings and desync all controllers, then enter on "Change controllers mode" menu and instead press "L and R" to connect, you should mantain the "SYNC" button on joycon's side for a few seconds and try again with "SYNC" button until controller appear connected on the screen.

When you back again to OFW gona have the same problem and you should do the same steps again to get joycon's connected.

This dosn't ocurrs before, it's a new problem but didn't know exactly when start, I see it like last month and today still.

milkdawg, 2021-11-17:
did you ever reformat your emummc?  If you ever did your bluetooth-id (or something like that is different).  A user has a tegraexplorer script for copying the file that contains bluetooth info from sys to emu.  https://gbatemp.net/threads/guide-pairing-the-joy-con-on-emunand-and-sysnand.562519/.  Sync all your controllers to sysmmc and then do this.  If you ever pair controller to different console will have to redo process.

DarkRaGaming, 2021-11-17:
> did you ever reformat your emummc?  If you ever did your bluetooth-id (or something like that is different).  A user has a tegraexplorer script for copying the file that contains bluetooth info from [...]

When I reinstall update I reinstall emummc

DarkRaGaming, 2021-11-17:
> Can confirm this bug, if you have paired the joycons on OFW and you switch to CFW, when you put out the joycons have problems to connect to Switch through BT.
> 
> To fix this, you should enter on system settings and desync all controllers, then enter on "Change controllers mode" menu and instead press "L and R" to connect, you should mantain the "SYNC" button on joycon's side for a few seconds and try again with "SYNC" button until controller appear connected on the screen.
> 
> When you back again to OFW gona have the same problem and you should do the same steps again to get joycon's connected.
> 
> I think it's related with joycon's FW, it's like they store some data that conflict with OFW and CFW (If it's console serial I have it faked on my CFW).
> 
> This dosn't ocurrs before, it's a new problem but didn't know exactly when start, I see it like last month and today still.
> 
> If you don't pair joycons on CFW and switch again to OFW the controllers work fine without any step, only get breaked on OFW when you pair them on CFW.
> 
> When you try to pair joycons with this bug, the animation of leds stop so fast (1 sec or minor) instead do for a few seconds like when they work without problems, if you try to use the "search controllers" on console the controllers appear with "?" simbol like they are not present or out of range.
> 
> Controllers work fine inserted in console, this issue appear only when you try to use their through BT.
> 
> One time paired through BT, my controllers didn't get disconnected.
> 
> OFW and CFW are on the lastest version (13.1.0), and the firmware of joycons are updated to lastest too, dosn't seems a version/firmware conflict.

Interesting so I was able to fix it with this. That it interesting. It happens to be every other update or few update after one.

closed 2025-04-03 by SciresM