Atmosphere-NX/Atmosphere #1707: panic occurred while running atmosphere

butawsky, 2021-11-12:
i randomly experience the error twice can we determine what is wrong using the bin file?

 im on 13.1.0 | ams 1.2.4

heres the bin file..

https://drive.google.com/file/d/1FH45Al7q5FI2hjFy44KcKGwKxvh19To6/view

thankyou very much..

Masamune3210, 2021-11-12:
Please go back and fill out the issue template instead of just deleting it

butawsky, 2021-11-12:
> Please go back and fill out the issue template instead of just deleting it

done and im sorry for my mistake

SciresM, 2021-11-15:
Looks like something killed the FS session being used for layeredfs mid-execution...I am just going to close this, as the most likely candidate is racey stuff that's not really fixable while doing mitm with no control over process termination.

If this triggers consistently, reach out to me on discord (`SciresM#0524`) and I'll poke at it.

Thank you for reporting!

closed 2021-11-15 by SciresM

butawsky, 2021-11-20:
thankyou very much too for the answer...