Atmosphere-NX/Atmosphere #1708: Switch getting slower

tsukina22, 2021-11-12:
## Bug Report

What's the issue you encountered?
Most of my games are opening more slower than usual, or when i start to play too, only Animal Crossing is working ok, with just a imperceptible lag, i don't know what happened, this never happend with me before, is but since Atmosphere 1.2.1 is happiness and now is getting worse. 

How can the issue be reproduced?
This happend mostly with the airplane mode on, with the wifi on, it gets better. 

Crash Report
There is no crash. 

System Firmware Version
13.1.0

Environment?
I'm using hekate 5.6.5
AMS 1.2.4

Homebrew software installed

Sys-clk
jksv
lan play
Edizon
ftpd
nxshell
nxtheme installer
Pplay
Tinfoil
amiibo
Tegra
Cheats - Update

EmuMMC or SysNAND:

emummc - partition base (i thought it was the emummc the problem, so i format everything and did a file based partition instead, but no use, the same problem) 

Additional information

I try to search on Reddit and Telegra and i found a lot of problems like that happening on 1.2.4.

fincs, 2021-11-12:
> Tinfoil

Piracy not supported. Also, you have way too many 3rd party things running in the background. Please try again with a fresh and clean copy of atmosphere straight from GitHub releases.

closed 2021-11-12 by fincs

tsukina22, 2021-11-12:
> > Tinfoil
> 
> Piracy not supported. Also, you have way too many 3rd party things running in the background. Please try again with a fresh and clean copy of atmosphere straight from GitHub releases.

I'm using Sysnand with my original games now, because o thought the emmumc was the problem, but is still opening my games shower, my original games! I think it shouldn't do that, even if I'm using cheats or amiibos.

Masamune3210, 2021-11-12:
Issue was already given, either you have too many things running at once or your sd card is slowing down due to age/use/just being a bad sd card