Atmosphere-NX/Atmosphere #1709: Fatal Error (Fusee-secondary.bin)

RoyallyOblivious, 2021-11-17:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ The Fatal Error code (stated above) showed up and I did everything to make atmosphere work again, but I keep getting that error]</br>

### How can the issue be reproduced?

[ * ]</br>
[ I downloaded the new firmware release here and unzipped the file into my SD card, then booted it with fusee, but it showed the error message and I didn't change anything internal in atmosphere. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version
13.1.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee
- Official release or unofficial build: Official Release
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: N/A [ * ]
- EmuMMC or SysNAND: (IDK)
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-11-17:
1) You didn't fill out the issue template

2) Update atmosphere, fusee-secondary doesn't exist for several months now.

closed 2021-11-17 by SciresM

RoyallyOblivious, 2021-11-17:
I did fill out the template, also I did update atmosphere, it still pops up

SciresM, 2021-11-17:
The template is not filled out.

i am quite sure your issue is a user problem -- this issue tracker is for bugs.

seek support in the Nintendo homebrew or ReSwitched discords.