Atmosphere-NX/Atmosphere #1710: FIFA 19 not working in Atmos 1.2.4

Luiscetina, 2021-11-17:
## **Bug Report**

### **What's the issue you encountered?**

When try to load a FIFA 19 game cartridge or in any other format in atmosphere 1.2.4.....an error ocurred and game close
The game can charge until black screen whith nintendo switch logos in the corners.... but after that... an error message is received while starting the game "the software was closed because an error occurred"

This error only show in recents release of atmosphere

the game not start in:

13.0 or 13.1 / atmos 1.0.0 or 1.1.1 or 1.2.0 or 1.2.1 or 1.2.4
12.1 / atmos 1.0.0 or 1.1.1 or 1.2.0 or 1.2.1 or 1.2.4

the game work on:

13.1 stock
12.1 / atmos 19.5
12.1 / atmos 20.1

other cartridge games work in any version of atmosphere... older and newer version of atmos..... only fifa 19 not work

### **How can the issue be reproduced?**

starting a fifa 19 original game cartridge or any other digital format

### **Crash Report**

there is not a crash report .... only an error message while starting FIFA 19 "the software was closed because an error occurred"

### **System Firmware Version**

13.1 , also tried on 12.1

### Environment?

Clean and fresh atmosphere new release whith newest hekate bottloader... and chainload fusee.bin payload
  - Clean atmosphere 1.2.4
  - Clean Hekate
  - All other  complementary files updated needed to work
- No sysmoduls installed
- Homebrew software installed:
  - JKSV
  - HEKATE TOOLBOX

- EmuMMC:
  - EmuMMC partition-based 

### Additional context?

tried on three diferent switch with same results

The game work with atmosphere 20.1 or lower....   the issues may be releated to the old file folder sept... but i not an expert 
all other game cartridge works fine 

Stratosphere configurated with NOGC = 0
I tried with airplane mode on and off

Sorry for my english is not my mother tongue

SciresM, 2021-11-17:
"the software was closed because an error occurred"

This means a crash report was generated.

Upload it.

Luiscetina, 2021-11-17:
> "the software was closed because an error occurred"
> 
> This means a crash report was generated.
> 
> Upload it.

sorry my bad..
[01637129791_0100ffa0093e8000.log](https://github.com/Atmosphere-NX/Atmosphere/files/7552241/01637129791_0100ffa0093e8000.log)
..

SciresM, 2021-11-17:
is fifa fucking serious?

```cpp

ASSERT(totalAvailableMemorySize > 3184_MB);
```

I don't actually know why we're failing that -- we shouldn't be altering the size of the application pool. I'll look into it.

SciresM, 2021-11-17:
@Luiscetina Please message me on discord (`SciresM#0524`) to test fix.

SciresM, 2021-11-17:
Fix confirmed by tester. Jesus christ @ 8 MB less reported available = game assertion crash.

commit 767e702a (2021-11-17): kern: correct TotalUserPhysicalMemorySize (closes #1710)
closed 2021-11-17 by SciresM