Atmosphere-NX/Atmosphere #1713: Panic occured when I try to boot Atmosphere after downloading BDSP

MewtwoFanz, 2021-11-20:
## Bug Report
N/A

### What's the issue you encountered?

I got brilliant diamond and shining pearl from gamestop and started playing them
I tried to edit some of the config files
I removed the ; 's from the system settings config files

### How can the issue be reproduced?

just boot into amosphere after installing updates and playing bdsp.

### Crash Report
I couldn't find one

### System Firmware Version

13.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: Official release version 1.2.4
- Do you have additional kips or sysmodules you're loading: yes
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: SysNAND

### Additional context?

SciresM, 2021-11-20:
If you're getting crashes, reports will be generated. Post screenshots and upload the reports.

MewtwoFanz, 2021-11-20:
[https://drive.google.com/file/d/1sx_ZcdbeZuz7tqxO1LRxWOIxsEJANvOu/view?usp=sharing](url)

I posted the report file here

MewtwoFanz, 2021-11-20:
I'll get a screenshot in a few minutes

MewtwoFanz, 2021-11-20:
[https://drive.google.com/file/d/1dFiHdF8Q8HX5Qh-SBXInhgiE6930cuiW/view?usp=sharing](url)

here's my screenshot of what i get when i try booting into atmosphere

MewtwoFanz, 2021-11-20:
Are these sufficient enough to see what's going on?

MewtwoFanz, 2021-11-20:
https://drive.google.com/file/d/1dFiHdF8Q8HX5Qh-SBXInhgiE6930cuiW/view?usp=sharing

https://drive.google.com/file/d/1sx_ZcdbeZuz7tqxO1LRxWOIxsEJANvOu/view?usp=sharing

Just copy these links into your browser, that's easier for me to do

Masamune3210, 2021-11-20:
You can paste pictures directly into this issue, instead of uploading them to drive

esppc, 2021-11-20:
I'm 99% sure that this is the result of a badly configured system_settings.ini

Configure it properly or just grab a fresh one from the samples

Yooooomi, 2021-11-20:
I'm having the exact same issue as the screenshot posted

cloud679, 2021-11-21:
why did you guys have to make a newer version of Atmosphere for

it broke my modded nintendo switch, when you guys released this newer version, the older version was not broken by any chance

cloud645, 2021-11-21:
nice try blocking me it won't work

cloud453, 2021-11-21:
I can do this forever

Yooooomi, 2021-11-21:
[report_00000000069414d8.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7576247/report_00000000069414d8.bin.zip)
Here is a report of the crash

SciresM, 2021-11-22:
Your system_settings.ini is in invalid format. Fix it up and you'll be fine @MewtwoFanz.

closed 2021-11-22 by SciresM