Atmosphere-NX/Atmosphere #1715: Error Code: 2168-0002 (0x4a8)

NaturalHarmonia, 2021-11-21:
## Bug Report

### What's the issue you encountered?

Pokemon Brilliant Diamond and Shining Pearl occasionally crashes when I'm using cheat codes. I've mostly been using the 60fps cheat. The Error code I get is 2168-0002 (0x4a8) Program: 420000000007E51A.
I haven't made any changes to Atmosphere beside from adding the necessary folders for cheats under the contents directory, changing ;dmnt_cheats_enabled_by_default = u8!0x1 to dmnt_cheats_enabled_by_default = u8!0x0 under system settings, and adding overlays and other things to use Tesla and EdiZon.

### How can the issue be reproduced?

The issue can be reproduced by often turning cheats on and off in rapid succession but also by normally playing the game with a cheat on.

### Crash Report

[Gist](https://gist.github.com/NaturalHarmonia/c77942bba562a26dae668888e2b9d6aa#file-01637423815_420000000007e51a-log)

### System Firmware Version

13.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee
- Official release or unofficial build: Official release Atmosphère 1.2.4
 - Do you have additional kips or sysmodules you're loading: Does nx-ovlloader count as a sysmodule? I'm not sure what a sysmodule is. Also I have sigpatches.
- Homebrew software installed: JKSV, AIO Switch Updater, Goldleaf, Daybreak, and others.
- EmuMMC or SysNAND: EmuMMC not sure if file or partition based because it's been a while since I hacked my switch.

### Additional context?

N/A

masagrator, 2021-11-21:
```
Also I have sigpatches.
```
Piracy is not supported.

SciresM, 2021-11-21:
Abort in Tesla is not an ams problem. Go talk to the Tesla developers.

closed 2021-11-21 by SciresM