Atmosphere-NX/Atmosphere #1716: Switch doesn't wake up when undocking

l-a-w-f-u-s, 2021-11-22:
## Bug Report

### What's the issue you encountered?

Switch doesn't automatically wake-up from sleepmode when undocking.

### How can the issue be reproduced?

- Put switch in sleepmode
- Undocking
- Switch doesn't automatically wake-up (Switch wake up and show lock screen on OFW 13.1.0)

### Crash Report

N/A, no crash on my switch

### System Firmware Version

12.1.0 on emuNand, 13.1.0 on sysNand

### Environment?

**Hekate v5.6.5
Atmosphere v1.2.4 w lastest sigpatches**

Homebrew app:
Checkpoint

No theme
No game mode installed

### Additional context?

N/A

SciresM, 2021-11-22:
1) is this actually a problem...? Does it occur without atmosphere?

2) post screenshot of atmosphere/contents? Does the issue occur without whatever's in there, if anything?

masagrator, 2021-11-22:
```
w lastest sigpatches
```
Ekhm...

Commander-Crash, 2021-11-22:
I have a issue like this but only CFW is shutdown. I would have to hold power bttn again for 10 seconds for it to power off.

artyq, 2021-11-25:
I have the same issue.  The switch doesn't wake up, so I need to hold the power button to shut it down fully.  

Hekate 5.6.5
Atmosphere 1.2.4
Emunand 13.1

SciresM, 2021-11-26:
I'm going to assume user error given non-communication and the fact that literally every time someone has reported this in the past it has been user error.

closed 2021-11-26 by SciresM