Atmosphere-NX/Atmosphere #1717: atmospherre error while loading modded game

Lioniox, 2021-11-25:
## Bug Report

I only recently started modifying the Switch.

When I started a with atmospherre modified game I got this error code.

Panic occured while running Atmospherre .

Title ID: 01007EF00011E000
Error: std : : abort (OxFFE)

I would look forward to some nice tips.

SciresM, 2021-11-25:
1. You didn't use the issue template
2. Upload the report file
3. Atmosphere fatal with title id = game should *never* happen. Indicates you fucked up somehow with your user contents.

SciresM, 2021-11-25:
Reopen using the issue templates.

closed 2021-11-25 by SciresM

Lioniox, 2021-11-26:
I will send it to morrow in my country its very late. My eyes are closing automaticly

Lioniox, 2021-11-26:
1. I cannot transfer the report file because it shows that it is damaged.
2. But the path is: atmosphere/fatal_errors/report_0a097eee.bin
3. Other games like Animal crossing new horizons work except Zelda botw that was the game I was having trouble with.