Atmosphere-NX/Atmosphere #1720: Ring Fit Adventure crashed while entering settings of the game

daniperez95, 2021-11-26:
## Bug Report

### What's the issue you encountered?

Ring Fit Adventure always crash when enter settings of the game.

### How can the issue be reproduced?

Open the game, select Adventure, press Y to go into Settings, wait or select anything and crash into atmosphere error screen.

### Crash Report

Is an atmosphere crash but there is nothing inside crash_reports

### System Firmware Version

12.0.3

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - Hekate 5.6.5
- Official release or unofficial build: 
  - Atmosphere 1.2.4-master-e256261b8
- EmuMMC or SysNAND:
  - SysNAND

### Additional context?

- Additional info about your environment:

Error Code: 2162-0002 (0x4a2)
Program: 010000000000100F
Firmware: 12.0.3 (Atmosphere 1.2.4-master-e256261b8)

Game: Ring Fit Adventure (1.2.0)

Other games work fine for this console, even this game if you dont enter in settings.

SciresM, 2021-11-26:
There should be a report inside fatal_reports. Can't really do anything without a report file.

What custom config do you have; if any? Able to post a screenshot of your atmosphere/contents folder?

daniperez95, 2021-11-26:
Oh yes, there were logs, i have 5 for the 5 crashes i got, all are the same, here is one https://gist.github.com/daniperez95/ea897ea3581620a923645f0df74336f5

And this is my contents folder
[image]

Thanks!

SciresM, 2021-11-26:
That is a *lot* of custom sysmodules, including Tesla which consumes applet memory at runtime.

Does the error go away if you remove custom sysmodules? My bet is that this isn't an ams issue, but that your custom sysmodules are eating up too much applet resources that ring fit can't spawn a web browser.

daniperez95, 2021-11-26:
Oh s*** i didnt think in that, so sorry, im closing this.

Really thank you!

closed 2021-11-26 by daniperez95

SciresM, 2021-11-26:
No worries, glad my intuition was right (I assume)!