Atmosphere-NX/Atmosphere #1721: atmospherre error while loading modded game

Lioniox, 2021-11-26:
## Bug Report

Panic occured while running Atmospherre .

Title ID: 01007EF00011E000
Error: std : : abort (OxFFE)

### What's the issue you encountered?

When I started a with atmospherre modified game I got this report: (look to Bug report

### How can the issue be reproduced?

Don't know it

### Crash Report

Crash reports can be found under /atmosphere/fatal_errors/report_0a0971eee.bin.
I can't tranfer it to my PC because it's saying that the file is damaged

### System Firmware Version

My version is 13.1.0

### Environment?

bootloader Atmosphère was launched by: fusee 
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Yes
- EmuMMC or SysNAND: EmuMMC
-If using an EmuMMC, include whether it's partition-based or file-based: Don't know it

### Additional context?

I was Just modding my Switch for a few days.

Masamune3210, 2021-11-26:
We need that crash report, if your computer wont let you copy it and is complaining about damage, your sd card is probably corrupted and might even be the root of your issues. Try fixing that

SciresM, 2021-11-26:
Re-open with the crash report attached.

I agree that if you can't access it, sounds like your SD card is fucked and not like there's an atmosphere issue.

closed 2021-11-26 by SciresM

Lioniox, 2021-11-26:
I can copy it now. It was a problem with my computer. here Is the file I have to put it in Zip to upload it on github

[report.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7610769/report.zip)

SciresM, 2021-11-27:
"0x815" error attempting to access btdrv from botw-title id.

Your SD card is fucked and/or your user-config is fucked, 1) botw should not be trying to create an atmosphere fatal error, 2) botw should not be trying to connect to btdrv.

I think you are trying to load a custom sysmodule (mission control?) overriding botw, somehow, just clean your atmosphere/contents dir.

Lioniox, 2021-11-27:
Ok I will Try it

Lioniox, 2021-11-27:
@SciresM ITS WORKING thank you so much!