Atmosphere-NX/Atmosphere #1722: HOS 13.0.0 crash on startup

hahnmt, 2021-11-26:
Bug Report

What's the issue you encountered?
After upgrading firmware to 13.0.0+exfat on HOS using daybreak, crashes on startup.  Trying to resurrect a HOS that was upgraded with a fat32 version, i performed an upgrade on fat32 card to add exfat drivers, then copying back file-based firmware to exfat card.  HOS launches on fat32 card, crashes on exfat card.  

A fatal error occurred when running atmosphere
Program ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

How can the issue be reproduced?
startup of HOS from atmosphere launch screen generates error after atmosphere logo

Crash Report

[report_0000000006e7c6ac.bin.txt](https://github.com/Atmosphere-NX/Atmosphere/files/7610520/report_0000000006e7c6ac.bin.txt)

System Firmware Version
13.1.0.  HOS 13.0.0

Environment?
hekate official 5.6.5. atmosphere 1.2.4
Homebrew software installed: most of the normal.  Tinfoil. Checkpoint. a pretty long list, no emuiibo or other apps, no custom themes
Emummc File based

SciresM, 2021-11-26:
ExFAT corrupts all the time. Not an ams problem.

closed 2021-11-26 by SciresM

hahnmt, 2021-11-26:
sounds great.  anything in the dump or is the path forward just "dont use exfat"?

takoyaki99, 2021-11-26:
don’t use exfat, exfat corrupts a lot

takoyaki99, 2021-11-26:
(is what the path forward should be)

SciresM, 2021-11-26:
Honestly the recommendation is literally just "don't use exFAT".

the report is an improperly encoded converted to txt, you would need to upload it as a bin for me to actually read it.

i expect it to indicate filesystem corruption though, if you really care I am willing to look at a proper bin upload to confirm.

hahnmt, 2021-11-26:
Thank you.  I would appreciate it if you could look over the bin dump while i prepare a fat32 sd for rebuild

[report_0000000006e7c6ac.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7610658/report_0000000006e7c6ac.zip)
7

SciresM, 2021-11-27:
fs::ResultInvalidFatFormatSd().

So...you don't have exFAT firmware installed. :P

hahnmt, 2021-11-27:
Oh man. This is killing me. 

I have upgraded HOS with daybreak 13.0.0 + exfat install. I am not joking. Literally before this crash I did another re-install.

rgarcia7, 2021-12-01:
To add to what SciresM and takoyaki mentioned. 
Fat 32 is the way to go. There is no benefit on using exfat, apart from having a time bomb waiting to destroy your files on the SD card.

More info at: https://gbatemp.net/threads/is-exfat-still-a-bad-idea.545779/post-8751906