Atmosphere-NX/Atmosphere #1725: Panic Occurred while running Atmosphére.

BlazeRevelation, 2021-11-29:
"Panic Occurred while running atmosphere

Title ID: 010041544d530000
Error:     std: :abort (0xFFE)

Report saved to /atmosphere/fatal_errors/report_054fbde6.bin

Press POWER to continue."

### What's the issue you encountered?

I have been using Atmosphere on my Switch without any issues recently.  Had just finished playing a game and went to archive it to save some space on the SD Card and received an error message from the Switch saying it couldn't access the SD Card and that I need to reinsert it and try again.  That error message was displayed for just a moment before I got a black and yellow error message from Atmosphere saying that Panic Occurred while running.  Now I am able to boot into Hekate, but cannot boot Atmosphere.  I continually get the error I typed out above.  I'm only creating this issue because I seem to have looked at all the other threads with this or a similar issue and nothing seems to work for me.  I've tried the Archive Bit Fixer, replacing the Atmosphere folder, deleting the Contents folder within Atmosphere and a few other things and I still get that same error.  The weird thing is, even though the Atmosphere error says that a report was saved to fatal_errors, that folder is completely empty every time.

### How can the issue be reproduced?

Boot into Hekate > Try booting Atmosphere > Get initial splash screen > Error appears

### Crash Report

There are no crash reports with the correct time stamp in any error report folder.

### System Firmware Version

12.1.0

### Environment?

Bootloader: Hekate 5.6.5

Atmosphere on SysNAND

### Additional context?

-Not using cheats, mods, or themes, only have Incognito.  I don't believe anything else that modifies anything crucial.
-On Windows

SciresM, 2021-11-29:
Upload the report file.

BlazeRevelation, 2021-11-29:
> Upload the report file.

There is no report file where it says the report has been saved.  Should I be doing something else or looking somewhere else than it says?

SciresM, 2021-11-29:
Trigger the report again. If that doesn't work, delete the fatal errors folder and recreate it, then trigger the error again.

Can't do anything without the report file.

BlazeRevelation, 2021-11-29:
> Trigger the report again. If that doesn't work, delete the fatal errors folder and recreate it, then trigger the error again.
> 
> Can't do anything without the report file.

Well, I just can't get the Report file to actually generate.  I've triggered it multiple times, deleted the fatal_errors folder, recreated, and nothing's there.  I get the same error every time, always tells me the report has been saved and the folder it empty every time.  Any other advice?

SciresM, 2021-11-29:
atmosphere dir might have archive bit set, use a different sd card, delete atmosphere folder/re-extract latest release, make sure you're not using macOS.

If you're using fusee, try using hekate.

If you're using hekate, try using fusee.

I literally cannot help you without the report file.

BlazeRevelation, 2021-11-29:
I understand and I appreciate you trying to help! Is there any known reason that an error report wouldn't generate?

SciresM, 2021-11-29:
SD card problem/archive bit, hence the suggestions I gave.

BlazeRevelation, 2021-11-29:
Alright, thank you!

closed 2021-11-29 by BlazeRevelation

Trixten, 2021-12-22:
Been getting the same error, anything I have from the fatal error log is in .bin version.  Any help would be appreciated

BlazeRevelation, 2021-12-23:
In my case, the card had permanently write-protected itself which prevented a crash log of any kind. I’m not sure if that was also the reason I was getting the error in the first place but I confirmed online the write-protection that can’t be turned off is a method to protect data as a last ditch effort by the card when it’s corrupted or failing. 

-I copied everything off the dead card, pasted it on desktop as backup, pasted it again on another card to test it and had no problems, concluded it was the card that was the problem, bought a new one and haven’t had a problem since then. 

Hope that helps!

kronictripy420, 2022-02-25:
Hello SciresM let me start by saying thanks for all of your hard work on this. ok i was using sx os 2.9.2 and OFW 9.0.1 no emunand or sysnand back up for the most part now there dead. is it ok to update OFW to 13.2.1 then put files from atmosphere and hekate, sigpatches on sd card and start emuMMC, which one is better sd partition or sd files. Well i just updated to OFW 13.2.1 put all files from Atmosphere, Hekate, and sigpatches, also found 2 sigpatches on another form that were not included in the gethub section for sigpatches. At first i couldnt get Atmosphere or Hekate, was just giving my Fatal error every time i tried to launch Atmosphere, because of the Hakate_ipl.ini, had to create it n paste in .txt and had to save it in different file extention (.ini) and now got Hekate working, now trying to load Atmosphere was getting Fatal Error, now im getting Panic Occured while Running Atmosphere and that all im getting could you help me please

masagrator, 2022-02-25:
> Hello SciresM let me start by saying thanks for all of your hard work on this. ok i was using sx os 2.9.2 and OFW 9.0.1 no emunand or sysnand back up for the most part now there dead. is it ok to [...]

1. Piracy is not supported.
2. You didn't use issue template.

kronictripy420, 2022-02-25:
im not piracy anthing, and how do i use issue template

kronictripy420, 2022-02-25:
@masagrator could you even help with my issue?

kronictripy420, 2022-02-25:
@masagrator plz help me if you can, im stuck in a paddle boat with no paddles

Masamune3210, 2022-02-25:
You were already told that piracy isnt supported. Sigpatches are considered piracy. Close this issue and either try to recreate the issue without the piracy means and fill out the template correctly or try to find help elseware

dmorrissette48, 2022-05-02:
@SciresM I have a similar issue as the person that made this post however im on MacOs. 

[report_048681cf.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8605859/report_048681cf.bin.zip)

Zoraa00, 2023-05-12:
FIX!!!

try this hekate, tools tab bottom right hand side click arch bit then fix bit

braxtron, 2023-11-17:
> FIX!!!
> 
> try this hekate, tools tab bottom right hand side click arch bit then fix bit

this worked for me. no clue how the error occurred in the first place. i had let the battery die and then it would no longer boot any payloads.

TEIXAREI44, 2025-02-18:
> FIX!!!
> 
> try this hekate, tools tab bottom right hand side click arch bit then fix bit

I tried this and it did not work I will attach the report if anyone could help me pls

[report_026ec2f83.txt](https://github.com/user-attachments/files/18855579/report_026ec2f83.txt)

impeeza, 2025-02-22:
> > FIX!!!
> > try this hekate, tools tab bottom right hand side click arch bit then fix bit
> 
> I tried this and it did not work I will attach the report if anyone could help me pls
> 
> [report_026ec2f83.txt](https://github.com/user-attachments/files/18855579/report_026ec2f83.txt)
> 
> I changed it to txt so that I could upload it

As told several times before, is a SD card problem, or is corrupt or have bad archive bits.  TRY A NEW CARD FORMATTED ON WINDOWS AND COPY THE FILES ON WINDOWS, THIS IS NOT A SUPPORT SITE, FOR SUPPORT, PLEAS GO TO DISCORD OF GBATEM.NET