Atmosphere-NX/Atmosphere #1726: Persistant black screen when using fusee-primary

TimeToFixStuff, 2021-12-01:
My Switch was working fine. Then I downloaded the tinfoil "self installer" and after running that, I can no longer get into CFW sys Atmosphere.

To be clear, what I see is the Atmosphere logo, and then a persistent black screen.

I can get into Hekate payload using my RCMloader, but not the fusee payload. Selecting "CFW (sysMMC)" in Hekate used to work fine, but now I get the same persistent black screen that I do when trying to use fusee payload. Selecting CFW emuMMC just complains with an error because I don't have that setup. Selecting Fusee from Hekate gives me the same black screen. Selecting Stock loads me into the stock firmware without CFW as expected.

I also tried to re-initialize from within the stock firmware so that the internal memory would be reset. That changed nothing. I have 3 different cards, all of them work, and I've tried both FAT32 and exFAT.

Any ideas? Any way I can get some debug out of this situation?

TimeToFixStuff, 2021-12-01:
I forgot to mention that I use sdsetup.com and it has always worked great.

masagrator, 2021-12-01:
1. You didn't use issue template
2. Piracy is not supported.

TimeToFixStuff, 2021-12-01:
Is homebrew supported?

masagrator, 2021-12-01:
Not installable ones

closed 2021-12-01 by SciresM