Atmosphere-NX/Atmosphere #1727: Always getting error 2168-0000 (0xa8) when open hbmenu (album)

yovin, 2021-12-02:


SciresM, 2021-12-02:
Upload the report.

fwiw -0000 indicates something more deeply wrong with your environment than usual.

yovin, 2021-12-03:
[image]

SciresM, 2021-12-03:
Upload the actual report file this generated, please? 

From atmosphere/crash_reports or atmosphere/fatal_reports.

yovin, 2021-12-04:
[01638548262_010000000000100d.log](https://github.com/Atmosphere-NX/Atmosphere/files/7653304/01638548262_010000000000100d.log)

yovin, 2021-12-04:
Sorry, this is the actual report. Thanks

MagnumFang, 2021-12-05:
I have this exact same issue

sanselon, 2021-12-06:
Hello, same here

SciresM, 2021-12-06:
This seems more than anything like your hbmenu nro is corrupted for some reason, tbh. Gonna tenatively close this, "undefined instruction" inside hbmenu is pretty nonsense/unreproducible locally (creport should be picking up hbmenu module name, too, not using a fallback build-id).

I recommend making sure your SD is formatted as FAT32 and re-extracting latest release zip, come to discord for further support if the problem isn't resolved by this.

closed 2021-12-06 by SciresM

yovin, 2021-12-07:
Can i get discord link?