Atmosphere-NX/Atmosphere #1728: Local multiplayer on certain games broken

RenoSuave, 2021-12-03:
## Bug Report

n/a

### What's the issue you encountered?

Local multiplayer for animal crossing new world results in communication error regardless of game version. Not sure if I've enabled some sort of setting to disable local multiplayer as it finds the village but instantly fails to connect. Was working on atmosphere 1.0 but since updating to 1.2.5 on firmware 13.1.0 the feature seams broken.

### How can the issue be reproduced?

Trying to connect 2 switchs on local multiplayer

### Crash Report

N/a

### System Firmware Version

13.1.0 U

### Environment?

- fusèe bootloader
- Official release version 1.2.5 
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: [ Goldleaf, tinfoil, Daybreak, retroarch, lockpick, reboot-to-payload ]
- SysNAND:

### Additional context?

I have tried on both Animal Crossing New Horizon version 2.0 as well as 1.1.0 and both have failed. I am not using idnmitm cfg. The 2 consoles do connect receiving a message "We found the island, loading now" following by "Preparing to board" Then instantly receiving " Communication error" then returning to the game. Any help would be much appreciated.

zemixe, 2021-12-03:
1. "tinfoil" piracy is not supported.
2. Sounds like https://github.com/Atmosphere-NX/Atmosphere/issues/1697 Again, piracy is not supported. Buy the game.

RenoSuave, 2021-12-03:
I have the game can post the 2 boxes if you would like. Tinfoil is a launch used to easily organize games and full sd card access all in one place if you didn't know as well as connect to retro arch for fast booting. Also unfortunately that is not the issue I am experiencing, if you would like to be helpful please reread

Masamune3210, 2021-12-03:
No, tinfoil is a piracy tool. Anything that requires sig patches is considered piracy and is not supported. Remove the sigpatches and make a new issue or modify this one once your environment is sane

RenoSuave, 2021-12-03:
I do not have sigpatchs installed. I did not realize this was such a toxic community and it is unfortunate.

RenoSuave, 2021-12-03:
and unless im mistaken, you would not need sigpatches for game carts, i could be wrong though

takoyaki99, 2021-12-03:
You have to have sigpatches for tinfoil to be installed.
Also, many of the devs are are anti-piracy.
Nintendo would not be happy if Atmosphere was pro-piracy (SX being pro-piracy sorta made them dissolve as a group), so it’s something needed for the continued survival of Switch homebrew.
Even if it seems annoying and “toxic”, it’s used to
make things around here work.
That’s just the baggage that comes with the Switch scene.

closed 2021-12-03 by SciresM