Atmosphere-NX/Atmosphere #1729: Error Code: 2144-0001 (0x290) Appears after update

Anakin072, 2021-12-03:
## Bug Report

Error Code: 2144-0001 (0x290) Will not let me launch any homebrew applications. I have tried to reinstall atmosphere, my sigpatches, specific homebrew applications that don't work to no avail.

### What's the issue you encountered?

I was attempting to update Pokemon Brilliant Diamond until Tinfoil failed to launch. I then noticed no other homebrew programs were working either.

### How can the issue be reproduced?

I updated my Switch firmware and my atmosphere and I can no longer do any homebrew-related tasks. It does boot into atmosphere though, just nothing homebrew related is usable.

### Crash Report

(https://gist.github.com/Anakin072/1a017065dbaf623f5b027e4f6f78dd32)

### System Firmware Version

1.3.2.0|AMS 1.2.5|S

### Environment?
[image]

I have been using the latest version of atmosphere, replacing the files and sigpatches don't see, to work.

### Additional context?
None of this happened before updating to 1.3.2.0, no apps installed through homebrew will launch or will say there is an error.

Masamune3210, 2021-12-03:
Piracy is not supported

SciresM, 2021-12-03:
1) Piracy/signature patches not supported.
2) 2345-0007 from hbloader means sd card corruption, not an atmosphere issue.

closed 2021-12-03 by SciresM

Anakin072, 2021-12-03:
> 

There is no piracy involved... I own Brilliant Diamond but cant update through nintendo online as my switch is banned.