Atmosphere-NX/Atmosphere #1731: 1) you didn't use the issue templates

Anakin072, 2021-12-03:
1) you didn't use the issue templates
2) seek further support in a discord server like nintendo-homebrew. This issue tracker is for bugs, not for support.

_Originally posted by @SciresM in https://github.com/Atmosphere-NX/Atmosphere/issues/1730#issuecomment-985253446_

I literally followed the template in my previous post that was marked as resolved, when it in fact is not. Can I be directed somewhere that will actually help me? The bug is my entire system failing out of the blue.

SciresM, 2021-12-03:
You are continuing to not use the issue templates.

This tracker not for support. It is for bugs. Please seek support in a discord server of your choice like N-H, use google

If you make another issue, I will block you from interacting with the organization.

SciresM, 2021-12-03:
And, again, to be clear -- this is a user problem/corruption. It is not a bug.

closed 2021-12-03 by SciresM