Atmosphere-NX/Atmosphere #1733: `fsmitm_redirect_saves_to_sd` is not working

rashevskyv, 2021-12-04:
## Bug Report
Save redirecting do not work if enabled. 

### What's the issue you encountered?

Tested with sysnand and emunand, game launched from cartridge. Tried with Witcher 3 and Sonic Forces. Game was launched, saved, then I searched for saves on `atmosphere` folder. According to atmosphere sources, there should be `atmosphere/saves` folder, but there was not.

`atmosphere\config\system_settings.ini`

```
[atmosphere]
enable_deprecated_hid_mitm = u8!0x1
dmnt_cheats_enabled_by_default = u8!0x0
fatal_auto_reboot_interval = u64!5000
fsmitm_redirect_saves_to_sd = u8!0x1

[contents_delivery]
enable_debug_api = u8!0x1

[usb]
; Enable USB 3.0 superspeed for homebrew
; 0 = USB 3.0 support is system default (usually disabled), 1 = USB 3.0 support is enabled.
usb30_force_enabled = u8!0x1
```

### System Firmware Version

13.2.0</br>

### Environment?
Boots with hekate and fss0. Official **Atmosphere 1.2.5** without any additional kips or modules

SciresM, 2021-12-04:
In addition to the global enable flag, you must add a content specific flag for each program you want to redirect saves for: https://github.com/Atmosphere-NX/Atmosphere/blob/e256261b809ff0371e67e6a11d47cf64867f3f4e/stratosphere/ams_mitm/source/fs_mitm/fs_mitm_service.cpp#L190

SciresM, 2021-12-04:
btw "enable_deprecated_hid_mitm" this was removed over a year ago and the setting does nothing.

closed 2021-12-04 by SciresM

RichardPaulAstley, 2024-07-07:
In case if it's necessary for the future, I just wanted to post a little 'how to' (shoutout to @Real96 for giving me a hand) : 

Just go to : /atmosphere/contents/'gameid'/flags/ (create any missing folder)
and add a blank "redirect_save.flag" file. It'll be done. You'll be able to find the saves inside the '/atmosphere/saves' folder.

Sorry for digging up that one but might be important to ressource this