Atmosphere-NX/Atmosphere #1734: Error Code: 2002-4364 (0x221802)

cam443, 2021-12-04:
## Bug Report

### What's the issue you encountered?
After updating atmosphere from 1.2.2 to 1.2.5, I am unable to boot into CFW without and error. [image]

### How can the issue be reproduced?
Replace old files with new atmosphere files and attempt to launch CFW.

### Crash Report
[01638649399_010000000000002e.log](https://github.com/Atmosphere-NX/Atmosphere/files/7654728/01638649399_010000000000002e.log)

### System Firmware Version

13.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: official
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: JKSV, Daybreak, Tinfoil, NXThemeInstaller, RetroArch, Checkpoint, a few more that I can't remember
- EmuMMC or SysNAND: Partition based EmuMMC

### Additional context?

I've seen some people saying updating while a theme is applied can break stuff. I had a theme applied when I updated. Not sure if this is related.

masagrator, 2021-12-04:
Parental Control service is throwing that save data is corrupted. 

Does going back to 1.2.2 solves the problem?

SciresM, 2021-12-04:
this is a save data corruption result. I would re-create nand (or delete the offending save file), see https://suchmememanyskill.github.io/guides/unbrick/ for help.

Not an atmosphere bug, so closing this -- seek support in a discord of your choice, rather than on the issue tracker.

closed 2021-12-04 by SciresM