Atmosphere-NX/Atmosphere #1736: Atmosphere Fatal Error when opening Smash

Navilence, 2021-12-05:
## Bug Report

A fatal error occurred when running atmosphere.
Program ID: 010041544d530000
Error Desc: std::abort{} called (0xffe)

Report saved to atmosphere/fatal_errors/report_000000009b5ae299.bin
Press POWER to reboot.

### What's the issue you encountered?

Atmosphere gives me this error when trying to open Smash Ultimate. I was playing modded Smash online just last night. I already had the latest update of Atmosphere when I was playing. Afterwards, I worked on adding more nus3audio files as music. I replaced a bunch of tracks through the custommusic/stream;/sound/bgm folders like I always did. I also edited the SmashMSBT files through Kuriimu to match the titles I replaced. I did also notice that I have another copy of the SmashMBST files in my Ultimate folder next to my Mod folder, which I don't think was needed so I deleted that. I imported the files wirelessly through sys-ftpd-light, and then tried opening Smash Ultimate again, where I was greeted with this error message. 

I can't comprehend what it is that I changed that caused this error. (From this point onward I made all changes directly instead of wirelessly.) I later downloaded the latest Arcropolis update since I apparently didn't have that yet (not the beta one), but the same issue persists. I tried redownloading the latest update of Atmosphere and the fusee bin again, which didn't work. I've tried actually removing my custom music folder, and removing the SmashMBST folder, but it still causes the error. To my knowledge sys-ftpd-light doesn't have a newer update, but it's the only thing left that I could understand would've caused a problem. 

### How can the issue be reproduced?

- Set Switch to RCM mode via jig
- Inject fusee.bin through TegraRcmGUI v2.6 portable
- Open Smash
- Game loads for like a minute or two
- The smash loading logo appears in the bottom right corner, and circles around for a bit before pausing, and then showing the above error message. 

Atmosphere loads into the home menu normally, including my theme, and other games are playable as normal. I can also still open the homebrew menu, though I haven't tried any of the homebrew menu's features recently. 

### Crash Report

It produces a fatal error report instead of a crash report. I'd share if needed, but the fatal error reports come in bin files, which it doesn't let me just drag and drop into a gist. If necessary, please explain how to share it. 

### System Firmware Version

13.2.0</br>

### Environment?

Launched with the latest fusee.bin file that comes with the official release of Atmosphere 1.2.5

sys-ftpd-light is the only program I have that I know is a sysmodule. 

There are 5 programs in my HB menu:
- Daybreak (I don't know what this does but I think it came with Atmosphere)
- HB App Store (I've browsed the store but never downloaded anything
- NXThemes Installer
- Reboot to Payload
- Themezer-NX

### Additional context?

I've heard themes can cause problems during updates, but I was able to play Smash just fine after updating Atmosphere and Fusee.bin, and my themes still appeared as normal after first updating Atmosphere as well.

My computer runs Windows 10, and the SD card I have is a SanDisk Ultra 128gb MicroSD card that I've been using since I got my switch.

Masamune3210, 2021-12-05:
You can zip the bin file and upload it iirc, might check for filesystem corruption, as sometimes ftpd can cause issues if for whatever reason something went wrong while writing or updating the files

SciresM, 2021-12-05:
Upload the report file. You cannot be helped without uploading it.

Navilence, 2021-12-05:
[report_000000009b5ae299.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7656920/report_000000009b5ae299.zip)
This should work I think.

Thanks for responses, I got work the rest of today, but I'll check back in tomorrow morning

SciresM, 2021-12-05:
fs::ResultFatFileSystemCorrupted().

If you're using exFAT, reformat as FAT32, this looks like it's just SD card corruption to me.

closed 2021-12-05 by SciresM