Atmosphere-NX/Atmosphere #1737: A fatal error occurred when running Atmosphere Program ID 010000000000002b Error Desc: std: :abort() (0xFFE)

Shinichisan, 2021-12-06:
## Bug Report

### What's the issue you encountered?

When trying to boot atmosphere by either using hekate or fuse, the color atmosphere screen is shown, then the black and white atmosphere screen, after that a black screen is shown for a minute more or less and after the black screen the following message is shown:

A fatal error occurred when running Atmosphere
Program ID 010000000000002b
Error Desc: std: :abort() (0xFFE) 

### How can the issue be reproduced?

The issue apears trying to boot atmosphere either with fusee or from hekate.

### Crash Report

[report_000000000ca7e8ca.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7660500/report_000000000ca7e8ca.zip)

### System Firmware Version

SysNand firmware version 1.13.0
EmuMMC firmware version 1.13.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
   Either fusee or hekate

- Official release or unofficial build: 
  Official release version 1.2.4 and 1.2.5 (I updated atmosphere thinking that perhaps an update could solve the issue)

- Do you have additional kips or sysmodules you're loading:
  Sys-clk, EdiZon and Emuiibo (but they were already installed more ago than a month before the crash and was working properly)
- Homebrew software installed:
  EdiZon, Emuiibo, Amiigo, Goldleaf, JKSV, NXThemesInstalle and PayloadLauncher (but they were already installed more ago than a month before the crash and was working properly)

- EmuMMC or SysNAND:
   EmuMMC partition-based

### Additional context?

I updated atmosphere and emuMMC firmware a week ago; I was using atmosphere 1.2.0 and firmware 1.3.0 (both sysNand and EmuMMC)

This crash only appears when trying to boot atmosphere; stock and adroid boot successfully.

I don't remember to have done anything outside normal usage before the crash appeared; it started to appear suddendly.

After the crash, I tried also to boot the same emuMMC using a fresh atmosphere 1.2.4 obtaining the same crash

switchgeek, 2021-12-06:
I'm having the same issue.
How to open the crash report binaries, which tool to use?

switchgeek, 2021-12-06:
One thing I have noticed is, when the console waits on standby some time (eg. on the dock) and if a game was open, home button does not work, it becomes impossible to go to the home screen. If a game was not open but console was waiting on the homescreen then if I open a game console crashes with the same error.

Shinichisan, 2021-12-06:
What you say about the games don't happen to me. The crash happens to me when entering atmosphere and I don't remember any strange behavior before the crash. Perhaps your issue is different to mine

SciresM, 2021-12-06:
this is a save data corruption result. I would re-create nand (or delete the offending save file), see https://suchmememanyskill.github.io/guides/unbrick/ for help.

Not an atmosphere bug, so closing this -- seek support in a discord of your choice, rather than on the issue tracker.

SciresM, 2021-12-06:
@switchgeek your issue is completely unrelated. Delete your custom sysmodules/theme.

closed 2021-12-06 by SciresM

switchgeek, 2021-12-06:
@SciresM Thanks, I don't have any sysmodules/theme. Can't boot with the same error message. Attaching my crash report [report_000000000bcb025f.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7661897/report_000000000bcb025f.zip)
b.

A fatal error occurred when running Atmosphere
Program ID 010000000000002

Can you also share a doc about how to read these error report binaries.

Thanks

Shinichisan, 2021-12-06:
@SciresM Thanks for your answer. Could you share a tool or doc to be able to read the binary report? I'd like to delete/fix that offending save file but I don't have a clue which's the save that causes the issue. Is the binary report indicating the save that causes the crash?

SciresM, 2021-12-06:
@switchgeek 

You have a corrupt PRODINFO/system save.

Follow the instructions/directions in #1256 for guidance. You'll need to restore a non-corrupt backup and/or delete the system settings savedata.

@Shinichisan There is no tools/docs really, I read them by eye in a hex editor. The report indicates the bad save file is 0x80000000000000D1.

switchgeek, 2021-12-06:
Thanks, will check that.

Shinichisan, 2021-12-06:
@SciresM Thank you very much for the help, I replaced the emuMMC save by the one of the sysNAND and it worked