Atmosphere-NX/Atmosphere #1738: A fatal error occurred when running Atmosphère

Norakchieth, 2021-12-07:
## Bug Report

A fatal error occurred when running Atmosphère.
Program ID: 0100415444530000
Error Desc: std::abort() called (Oxffe)

Report saved to /atmosphere/fatal _errors/report_00000000076e0Zaf.bin
Press POWER to reboot.

### What's the issue you encountered?

I was attempting to copy game files from PC to SD card as usual. then I accidentally  open tinfoil and then I closed it right away. one more thing I installing some games after installation complete, it crushed then restart again. I really don't know which case that let this error occurred. 

### How can the issue be reproduced?

I just updated my Switch firmware from 10.2,0 to 13.0.0 and my atmosphere. after I used payload to start atmosphere.bin and then it splash atmosphere icon and then it's start to pop up the problem. 

### Crash Report

https://drive.google.com/file/d/1P0D0u1kb0SaEuTrElxEYdhRuY3TCw2hw/view?usp=sharing

### System Firmware Version

System Firmware 13.0.0 and atmosphere. 

### Environment?

I use hekate. 

### Additional context?

I just update from SXOS to atmosphere. Now I mess up. Please help. 

- Additional info about your environment:
- [ Any other information relevant to your issue. ]
[report_00000000076e02af.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7664730/report_00000000076e02af.zip)

SciresM, 2021-12-07:
Piracy is not supported.

closed 2021-12-07 by SciresM

Norakchieth, 2021-12-07:
Could you please explain me more about this error? I'm new to this and what should I do next? Please give advise sir.

Masamune3210, 2021-12-07:
Remove the signature patches and any piracy tools and then make a new issue.