Atmosphere-NX/Atmosphere #1744: fusee.bin not working

jojopipe, 2021-12-18:
I updated my switch a couple days ago and to make sure it works I also downloaded a new atmosphere version, but after updating atmosphere booting via the fusee.bin (ver. 1.2.5) just resulted in a black screen for me. I tried using an older version (1.2.4) which, to my surprise, worked on the first attempt.
Apparently there's something wrong with version 1.2.5, please look into that.

closed 2021-12-18 by jojopipe

masagrator, 2021-12-18:
You didn't use issue template, and didn't include anything helpful. 
It works for everybody involved in project, so how devs should look for any issue if they don't see one?
And why did you close issue right after making it?

SciresM, 2021-12-19:
Yeah, got didn't use the issue templates, and also 1.2.5 is fine. The problem is with your setup, not ams.