Atmosphere-NX/Atmosphere #1746: panic occured while running atmosphere. title ID: 010000000000002B

MauricioAdolfoIlabacaSandoval, 2021-12-24:
### What is the problem you found?

A backup was made of the atmosfered folder and the botloader which, at the time of starting the nintendo throws me the error mentioned in the pos, I tried to use a previous backup of the same folder and nothing, the message appears in letters assuming yellow that it is the latest version, perform the atmospheric discharge and place them in the sd, having the same error </br>
[Did you make any changes related to Atmosphère himself? ] </br>
 try to put the most recent version of the program, even if I remove the sd several times without having turned off the console, adenas every time I copied a file from mac I get strange characters when eating the name ".-" </br>

### System firmware version

1.25 </br>
[Replace Xs with the system firmware version at the time of crash. ] </br>
[You can find your firmware version in Settings -> System, under "System update". ] </br>
[If it says "Pending update", you can clear the pending update by restarting to maintenance mode. ] </br>

### Atmosphere?

- Atmosphère fuse for:
- Official launch:
  - [Official release version 1.25 (or) unofficial compilation]
      - [Ex: autocompilation]
     - [Ex: distribution of Atmosphère by Kosmos]
- Do you have additional kips or system modules that you are charging?
- Homebrew software installed: [*]
- EmuMMC or SysNAND:
  - [If using an EmuMMC, include if it is partition or file based. ]

### Additional context?

- Additional information about your environment:
It all started because I wanted to install the zelda btw in which, I couldn't because the awoo installer crashed, giving an error in atmosfered, I tried some ways with the R button, however the same problem, so I thought it could be a version problem, but when I update the atmospheric folder from a mac, the files were left plus other files with strange names, which, I saw that it did not work, copy the files from a windows pc.

which problem persists

takoyaki99, 2021-12-24:
Your atmos files might be corrupted.
Try using a fresh copy of atmosphere, and also try fixing the archive bit in Hekate settings.

(Also, the devs here don’t like any discussion of game backups/piracy/sigpatches, so make sure not to talk about them here. Or else you may not get help (for legal-ish reasons I think).

SciresM, 2022-01-03:
Closing because this is the first I'm seeing this issue in two weeks and the user didn't actually fill out the issue templates properly, didn't upload crash reports, and is a pirate, which is not supported.

closed 2022-01-03 by SciresM