Atmosphere-NX/Atmosphere #1747: Issue updating switch firmware

cmilhoan7, 2021-12-26:
### What's the issue you encountered?

I have been using a 16 GB micro SD card and recently received an 128 GB one. I needed to update my switch to be able to use exFAT micro SDcards, so I followed the steps on switch.homebrew.guide for updating a switch and selected FAT32+exFAT. I then powered the switch off and rebooted through RCM and hekate. I then recieved the error;

"Panic occurred while running Atmosphere. 
Title ID: 010...02B
Error: std::abort (0XFFE)"

Upon doing some research online and on github, I _think_ that I have accidentally used a bad firmware file somehow, or my switch is otherwise corrupted now. I hope I'm wrong or I hope that someone can point me in the right direction. Any help is greatly appreciated. I'm also new to the homebrew scene etc. and github itself so I hope that you'll pardon my ignorance. 

### How can the issue be reproduced?

I launched the payloads Fusee_atmosphere.bin and hekate_ctcaer_5.0.0.bin, both times I received the error message above and then I powered off the system.

### Crash Report

N/a

### System Firmware Version

13.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:

- Official release or unofficial build: 
- I launched the payload Fusee_atmosphere.bin and hekate_ctcaer_5.0.0.bin. I'm pretty sure I got both from the switch.homebrew.guide guide
- Do you have additional kips or sysmodules you're loading:
- I am unfamiliar with both "kips" and "sysmodules," so most likely not.
- Homebrew software installed: [ * ]
- Just what is recommended by the aforementioned guide.
- EmuMMC or SysNAND:
  - I did not set up an EmuMMC, I was going to do that on the new 128 GB card. I don't fully understand what either are, I know I didn't set up an emuMMC. I'm running thing on my switch that was bought on day 1 so it should be safe on the latest firmware and everything. 

### Additional context?

I think I have provided everything that is relevant but please let me know if you need any additional context from me.

takoyaki99, 2021-12-26:
I’d check and see if the 010...02B file (whatever the full title ID is) is in your SD’s atmosphere/contents directory. It’s happened to me a few different times and I’ve been panicking about a brick each time, but it was an outdated homebrew thing (usually a plugin).
This is more of a last-resort option, but:
Maybe try restoring a NAND backup (basically a prior snapshot of your system), if you have one. Your installed games and save data will be reverted to the point in time when you hopefully made it.
My recommendation is to just look for other answers too, and research!
Also, just for future use, FAT32 is recommended for homebrew use, also (exFAT corrupts easily on Switch).
I’m not a dev so I can’t tell you what any error codes mean, but it’s good that you researched.
Did you get the firmware files from DarthSternie (and updating with Daybreak) or from Nintendo officially (or from somewhere else?). I dunno if it’d make a difference

takoyaki99, 2021-12-26:
I also just saw: you’re using an outdated version of Hekate! Update your Hekate and Atmosphere folders from 5.0.0 for Hekate and whatever your Atmos is to both their latest releases (found on their Githubs)
Not saying that’ll solve it but, who knows! Except for the devs, lol

cmilhoan7, 2021-12-26:
Thank you so much for the fast response!

-I'll check to see if that file is somewhere in the directory and get back to you on that. I put "..." because it was a bunch of zeros, wasn't sure if it was necessary to count how many. If it is then I'll boot it up again, count, and update the post. 
-I followed the backup instructions on that guide so I should be able to do that if nothing else works. 
-Do you know if there's a reliable way to convert the sd card then? (lexar 128 gb if it matters) I've seen that people have had issues with converting bigger sd cards but maybe I didn't look hard enough.
-I did get the firmware files from DeathSternie, the latest ones. I followed the instructions to a T and got the files from there.
-Good call on the updating hekate and atmosphere I'll try that too.

PS: I love takoyaki, great username. Can't find much of it in my state

takoyaki99, 2021-12-26:
Gyah thank you so much for your kindness~!

As for file checking, sounds good! Try that!

Glad you have a NAND backup. You’re covered.

As for SD conversion, Hekate has a tool to convert to Fat32. I use it because I don’t have a (reliable) computer. I actually used it today to convert my new Wii U SD to Fat32, so yeah, it’ll work well.
I’ve converted 400GB and 200GB successfully with Hekate.

Nice job for getting stuff from DarthSternie. I’ve used updates from them and have been pretty fine, but sometimes crazy issues happen…but that’s what comes with learning about console hacking!

(Thanks for the compliment btw! I’m lucky because in my area there’s a good ramen-ya with takoyaki. My family would hate me making it, but I’ve wanted to try it for a looooong time)

cmilhoan7, 2021-12-26:
@takoyaki99 

Some good and bad news. 

-First of all, updating hekate and atmosphere lead to me no longer receiving the "panic" error when I boot hekate. However, when I boot to sysMMC it now gives a fatal error and crash of the same title and error code as the OG post, but it can not be written to the SD card, so there's no way to post the new crash log because it can't write it. 
    I tried booting it up with my old 16 gb card with everything newly updated and nothing bad happened and everything booted as expected. So, for some reason, even after updating to read exFAT, my exFAT card is still not being read/written properly. 
- This seems like good news I think because it means my switch is A-OK but my card is still giving me issues. 
- Since I can now boot into hekate but not into the actual CFW sysMMC with the 128GB Lexar, I can try formatting the SD card and then just dragging and dropping the stuff from my 16 gb card over again but this time my card will be formatted as FAT32 instead of exFAT.
    - could you help with that? I think I go to partition manager and just keep the default option (HOS (FAT32) and tap next step? I don't want to do anything irreversible so I want to hear it from you since you clearly know what you are doing.
    - I tried to add an image but like I said I've never used github so not sure if it'll work or if it's just a link
[image]
 
Thanks again so much for your patience and help

takoyaki99, 2021-12-26:
I’ve heard about this type of error before!
I think it’s caused by improperly formatted stuff? I forget why it happens (the explanation went waaaay over my head), but I do think that formatting will definitely help.
Make sure you have a backup of your SD’s files!!
I can see the image just OK~
So, if you don’t want an emuMMC you’ll want to partition all to HOS (Short for Horizon, which is the Switch’s operating system). The other options are good if you want to do stuff with either SwitchRoot Android or L4T Ubuntu on your Switch.
If you don’t want an emuMMC do it all to HOS.
As for filesizes, I use a 400GB SD, and don’t use emuMMC, and have a 15GB (IIRC) partition for L4T.

Basically, pick your emuMMC filesize (you can install to the emuMMC system memory, which is still part of your SD card!), and then go for it!
You need at least 2GB for the operating system to be set up. Small emuMMC used to run slow (when I used a 200GB I had a 5GB emuMMC and it ran slow, so I never used it), but I don’t think you’ll have problems.

In the end—it depends on if you want emuMMC or not, but, you’re doing great~

cmilhoan7, 2021-12-26:
If I do just HOS and emuMMC, can set up a partition for linux later or do I have to do everything on this first time I partition it?
I want to do Linux eventually but for right now I'm just worried about getting this SD card into a state of use and to boot up without issues. I'm hoping I can just go into this same menu in a few days and set up a linux partition on top of what would already be formatted.

cmilhoan7, 2021-12-26:
@takoyaki99 I think what I mean is; Am I allowed to make a partition on the same SD card as many times as I want?

takoyaki99, 2021-12-26:
Nope, you have to reformat it every time, which sucks.
I’d suggest using either another SD for linux (even though that’s super tedious), or format once and hope for the best!

cmilhoan7, 2021-12-26:
You have been a massive help thank you so much, I'll respond soon with an update. Hekate is creating the emuMMC, I'll have to restore my backup on the card then see if I get any errors. I'll post soon. I can't believe it's been two hours it's flown by. 4:00 AM for me

cmilhoan7, 2021-12-26:
I'll have to post in a few hours after I sleep, it's taking longer than expected to put the files on the sd card. Be back in about 6-7 hours

SciresM, 2021-12-26:
1) you didn't upload the relevant error report, and official support can't be provided without it.

2) the issue tracker is for bugs, not support. This kind of thing should be on discord instead.

closed 2021-12-26 by SciresM

cmilhoan7, 2021-12-26:
Okay @SciresM @takoyaki99 is there a discord server that you recommend for support then?  I wasn't sure where else to turn since I don't have enough reddit karma yet to post anywhere or do anything, I thought this would be a good spot to get help. Sorry guys, I'll join the discord server you point me towards. Like I said I'm new to this, didn't mean to waste anyone's time or the space here on the bug tracker.

cmilhoan7, 2021-12-26:
Team Neptune?

cmilhoan7, 2021-12-26:
Sorry I didn't upload a relevant error report. Atmosphere was failing to write the report to my SD card, which was another thing I was hoping to resolve

takoyaki99, 2021-12-26:
I’ve heard the Switchroot discord is good? I think the switch.hacks.guide team has a discord too.

I hope you get help!!!

cmilhoan7, 2021-12-26:
You're an absolute lifesaver Takoyaki thank you so much for all your help!