Atmosphere-NX/Atmosphere #1749: Blackscreen on 'Nintendo Switch' logo after i rebooted my console on the exception handler screen

DevCoud18, 2021-12-28:
## Bug

Blackscreen on 'Nintendo Switch' logo after i rebooted my console on the exception handler screen

### What's the issue you encountered?
I got an blackscreen in emummc on 'Nintendo Switch' logo after i have rebooted my console by the exception handler screen

### How can the issue be reproduced?

I have tested some homebrew apps on 13.X emummc to make sure witch homebrews needs to be updated. In this progress i got some crash screens by atmosphere but after one reboot, my console stucks on boot after get the 'Nintendo Switch‘ logo. I have looked whats the issue but i don‘t know why the exception handler broke my emummc. I have recreated my emummc... 4.1.0 booting fine but after reupdating to 13.X my console stucks on boot. I have atmo updated via AIO updater... no outdated sysmodules installed. SD Card works fine in fat32...

### Crash reports

N/A

### System Firmware Version

Sys 4.1.0
Emu 13.2.0 </br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fuse.bin
- Official release or unofficial build: official v1.2.5
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Only nro's
- EmuMMC or SysNAND: EmuMMC as SD Partition

### Additional context?
I don‘t know what i doed or doing wrong. I have only tested my homebrew apps witch one was outdated. No CFW or EmuMMC changes

SciresM, 2021-12-28:
Upload the relevant fatal error report.

SciresM, 2021-12-28:
Also it's probably sd card corruption.

DevCoud18, 2021-12-29:
> Upload the relevant fatal error report.

Here's my latest dump i got
[01640508237_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/7787936/01640508237_0100000000001000.log)
i pressed the power button to reboot into atmosphere but then i stuck after 'Nintendo Switch' logo

DevCoud18, 2021-12-29:
> Also it's probably sd card corruption.

I don‘t thing thats the problem because i used an other sd card but these don‘t work too

Masamune3210, 2021-12-29:
If you just copied the files from one card to another the corruption would still likely be present in the files

DevCoud18, 2021-12-29:
> If you just copied the files from one card to another the corruption would still likely be present in the files

That's not true. I have used clean downloads from the web.

Masamune3210, 2021-12-30:
Notice the first word in that sentence.....

SciresM, 2022-01-03:
If the crashing thing is qlaunch, this is not an atmosphere issue. Seek support on discord, probably your custom theme is bad.

closed 2022-01-03 by SciresM

DevCoud18, 2022-01-05:
> If the crashing thing is qlaunch, this is not an atmosphere issue. Seek support on discord, probably your custom theme is bad.

No, this isn't an custom theme issue. I don't have installed an theme. And it‘s an atmo issue because i got this dump serial times who i tested my apps and after one, I suddenly got an blackscreen after i rebooted from this exception handler screen. And only the emummc on 13.X is affected. Sysmmc works on 4.1. Where is there, please not an atmo bug? And please don't come with this Discord server. Unfortunately, it is useless considering that I have already reported tens of errors there, but it is never responded to no matter what I write in there. I see you didn't even read the message properly because it's about a black screen that suddenly occurred and not about the error dump that I received tens of times before, where the switch still worked perfectly. It's just only about the black screen that occurred where I rebooted from one of your error screens. No theme and the SD works. So it's a bug at Atmosphere.

Masamune3210, 2022-01-06:
You know, you get more flies with honey than with wine.....