Atmosphere-NX/Atmosphere #1750: Overlayfs for exefs crashes instantly

martmists-gh, 2021-12-29:
## Bug Report

### What's the issue you encountered?

While developing mods, I often need to add an extra exefs file (e.g. subsdk1) that didn't exist in the exefs. However, putting these files in the exefs directory crashes the game instantly with "Could not start the software. Please try again from the HOME Menu".

### How can the issue be reproduced?

create a file that didn't originally exist in exefs and put it in the exefs overlayfs directory.

### Crash Report

N/A

### System Firmware Version

13.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 1.2.5
- Do you have additional kips or sysmodules you're loading: sys-ftpd-light, also happens without
- Homebrew software installed: Amiigo, Checkpoint, Goldleaf, HB store, nxdumptool, nxgallery, nxshell, AIO switch updater, Daybreak, IPSwitch
- EmuMMC or SysNAND: Partition-based EmuMMC

### Additional context?

N/A

masagrator, 2021-12-29:
This works without issue with skyline plugin.
Issue is not with atmosphere but with your attempt to try link something that breaks game.

martmists-gh, 2021-12-29:
> Issue is not with atmosphere but with your attempt to try link something that is not adjusted to this game.

I'm trying to load it so I can see where it gets placed for use with Starlight, as documented [here](https://github.com/brycewithfiveunderscores/Starlight-SMO-Example/pull/1) which states any valid subsdk1 should work just fine. However, the game crashes instantly so the module doesn't get loaded.

martmists-gh, 2021-12-29:
Turned out to be a filezilla issue with newlines

closed 2021-12-29 by martmists-gh

breadbored, 2022-07-03:
@Martmists-GH Can you expand on how FileZilla caused this? I'm having the exact same issue and I'm using FileZilla, however I'm also running `sendPatch.py` so the mod is uploaded via `ftplib`. What files were FileZilla causing issues with?

Martmists-GH, 2022-07-03:
FileZilla's *default* behavior is to transfer extension-less files (like `subsdk1`) as text, replacing `\n` with `\r\n` which of course makes the game crash.