Atmosphere-NX/Atmosphere #1753: A fatal error occured wen running Atmosphere

tantoinet, 2022-01-02:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Old, unknown, Atmosphere version was correctly running (potentially on OFW 11.x).
I've turned off the console, FAT32 formatted the MicroSD card and put the new Atmosphere 1.2.5.
Booting using the latest fusee.bin results in an error message saying:
A fatal error occured when running Atmosphere.
Program ID: 0100000000000005
Error Desc: std::abort() called (0xffe)

### How can the issue be reproduced?

Not sure, but this might be related to FW v11.x. The console should have been turned off properly.
The MicroSD card (in its adapter) was written using Win11 + guiformat-x64 v1.01.

### Crash Report

[Crash reports](https://www.mediafire.com/file/w4sy72f4r5doqo6/fatal_errors.zip/file)

### System Firmware Version

Probably 11.x

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: tried both fusee and hekate.
- Official release or unofficial build: official 1.2.5
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: none
- EmuMMC or SysNAND: SysNAND

dialga5, 2022-01-02:
Hello, I was also having an issue with the same description outside of the ID being 0100000000000008, have you tried editing the hekate.ipl to use fusee.bin instead of fusee-primary.bin? 
so it would read out as:
Payload=bootloader/payloads/fusee.bin

SciresM, 2022-01-03:
@tantoinet TargetNotFound when trying to mount system version title system data from built-in-system. I'm going to hazard a guess that you have an improper update installed (choi-nx?)

At any rate, I would follow https://suchmememanyskill.github.io/guides/unbrick/ to get a known-good system partition, since this isn't an atmosphere bug.

@dialga5 if the title id is different, your issue is completely different. This issue tracker is for issues, not for support -- seek help on discord.

closed 2022-01-03 by SciresM

tantoinet, 2022-01-03:
thanks, restoring a backup of eMMC BOOT0, BOOT1 and eMMC RAW GPP fixed it