Atmosphere-NX/Atmosphere #1754: Fatal Error occured, Abort called lr=0x40011825

xemnass-xehanort, 2022-01-03:
## Bug Report

### What's the issue you encountered?

Months ago, during the summer, it suddenly froze and then was unable to boot properly... i did wat i though was right and restore to factory ofw with the nintendo "safe mode" loosing then the ability to use RCM ( it is an unpatched switch but i lost my RCM jig ) and still no boot. Then, by it self, a day light up and let me use it again so... again used atmosphere etc..... this time i tought to give a try to the emuMMC. Then, on december, happened again ( from inside the emuMMC ) but even the untouched stock ofw would not boot. Every tries led  to kernel panic, messed up splash screen ( all green glitch ) and 0xF00 errors. 
I didn't make any changes to Atmosphere.

### How can the issue be reproduced?

### Crash Report
[repo.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7803417/repo.zip)

### System Firmware Version

12.0.2
### Environment?

- Hekate
- Official release or unofficial build: 
  - latest release
- EmuMMC and SysNAND:

### Additional context?
Reading the crash report i could see the TLS Dump. It is all populated, as you can see by your self, by "cc" bytes and stragely the pstate pointer points to 0xcccccc. I don't know if something has happened to the switch. I even replaced the emmc physically with 2 other donor chips, restored a good backup of BOOT0/1 and full raw nand and evene rebuilt the firmware from scratch with HacGen and flashed through the use of BalenaEtcher/HacDiskMount and TegraExplorer script.

I'm willing to debug and look into it deeper but i wasn't able to find any info on how to debug the boot process. That would be really usefull. Neither was i able to find any special payloads to test the switch components in terms of RAM / CPU / EMMC etc....

DavidBuchanan314, 2022-01-03:
This sounds like a hardware issue, not an atmosphere issue.

xemnass-xehanort, 2022-01-03:
if that so, how can i test it out? is there anything i can do? if i could debug the boot process, i could see where it panics and the hardware part responsible. (e.g. if it expected certain values in the registers, or decryption of BOOT0 issues etc....)

SciresM, 2022-01-03:
The report you uploaded isn't the one you took a picture of.

It's mesosphere reporting invalid/malformed initial process binary (and the report is generated with ams 1.2.4, not 1.2.5). This usually happens when hardware problem or invalid/malformed boot0/package2 partitions.

You mention specifically restoring known good partitions...so I'm not really sure what kind of debugging you can do? I'm relatively confident this isn't actually a software problem, but.

Want to upload the actual fusee report?

xemnass-xehanort, 2022-01-03:
Yes i can provide anything you ask so long as i know where it is stored >.< i found this report under atmosphere/fatal_errors booting with package3 (that show a blank, black, screen). When i run with fuseè it doesn't generate a report, should i add something to the hekate config?

(For references:)
[config]
autoboot=0
autoboot_list=0
autohosoff=1
autonogc=1
backlight=100
bootwait=3
updater2p=0

[CFW (sysMMC)]  <---- black screen and report generation
emummc_force_disable=1
fss0=atmosphere/package3
icon=bootloader/res/icon_payload.bmp

[CFW (emuMMC)] <--- N/A
emummcforce=1
fss0=atmosphere/package3
icon=bootloader/res/icon_payload.bmp

[Stock] <---- black screen and no report generation
emummc_force_disable=1
fss0=atmosphere/package3
icon=bootloader/res/icon_switch.bmp
stock=1

[Fusee] <----- glitched screen but i don't see any generated report
icon=bootloader/res/icon_payload.bmp
payload=bootloader/payloads/fusee.bin

xemnass-xehanort, 2022-01-03:
(RIght now, out of pure randomness, and out of months of trials, but with dried out battery, seems to have boot up, but then shutdown due to low battery)

Now, completely out of any possible guessing.... Can it be a battery releated problem? Like wrong voltage?
Both of the cases when all of this happened ( the original freze that lead up to the brick ) happened after i left charging some days on the original docking station.

SciresM, 2022-01-03:
I have no idea. I am pretty confident this isn't a code issue though, so closing this -- feel free to hop on discord to chat for more support as needed?

closed 2022-01-03 by SciresM