Atmosphere-NX/Atmosphere #1757: Atmosphere error

Darkdragon20101, 2022-01-08:
I have an error with atmosphere, when i open smash ultimate and comes out a pop-up and i press continue, i have this error, same with super mario odyssey in the guide and controls, sorry for my english, i will wait for an answer, thanks in advance.
Here i post my error:

Atmosphère Fatal Report (v1.1):
Result:                          0x4A2 (2162-0002)

Program ID:                      010000000000100f
Process Name:                    LibAppletOf
Firmware:                        13.2.0 (Atmosphère 1.2.5-master-1019bc54e)
General Purpose Registers:
         FP:                     00000004d428ab80
         LR:                     000000457f275c84
         SP:                     00000004d428ab80
         PC:                     000000457f28d8f8
Start Address:                   000000457ec00000
Stack Trace:
        ReturnAddress[00]:       000000457eff3c8c
        ReturnAddress[01]:       000000457eff3a58
        ReturnAddress[02]:       000000457eff3634
        ReturnAddress[03]:       000000457eff3788
        ReturnAddress[04]:       000000457eea91ac
        ReturnAddress[05]:       000000457eea8390
        ReturnAddress[06]:       000000457eea84f8
        ReturnAddress[07]:       000000457edb2024
        ReturnAddress[08]:       000000457edb2ae4
        ReturnAddress[09]:       000000457edb25e8
        ReturnAddress[10]:       000000457eccb4bc
        ReturnAddress[11]:       000000457eccef90
        ReturnAddress[12]:       000000457edd66d0
        ReturnAddress[13]:       000000457f25ef38
        ReturnAddress[14]:       000000457f263534
        ReturnAddress[15]:       0000000000000000
Stack Dump:                               00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             0004d428ab80 20 ac 28 d4 04 00 00 00 8c 3c ff 7e 45 00 00 00
                             0004d428ab90 01 01 00 00 00 00 10 00 0c 00 00 80 01 00 00 00
                             0004d428aba0 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0004d428abb0 05 00 00 00 00 00 00 00 00 00 00 00 08 00 00 00
                             0004d428abc0 00 00 00 00 14 00 00 00 c0 64 27 7f 01 00 00 00
                             0004d428abd0 d8 ab 28 d4 04 00 00 00 00 00 00 00 00 00 00 00
                             0004d428abe0 00 90 7c 98 04 00 00 00 00 00 00 00 00 00 00 00
                             0004d428abf0 00 00 00 00 00 00 00 00 15 00 00 00 00 00 00 00
                             0004d428ac00 30 b8 74 7f 45 00 00 00 50 02 00 00 00 00 00 00
                             0004d428ac10 90 ab 28 d4 04 00 00 00 00 ae 6d 01 35 00 00 00
                             0004d428ac20 70 ac 28 d4 04 00 00 00 58 3a ff 7e 45 00 00 00
                             0004d428ac30 f8 84 ea 7e 45 00 00 00 08 ae 28 d4 04 00 00 00
                             0004d428ac40 d5 d8 43 7f 45 00 00 00 00 00 00 00 00 00 00 00
                             0004d428ac50 e0 89 66 7f 45 00 00 00 d8 ac 28 d4 04 00 00 00
                             0004d428ac60 00 00 00 00 00 00 00 00 d0 88 66 7f 45 00 00 00
                             0004d428ac70 a0 ac 28 d4 04 00 00 00 34 36 ff 7e 45 00 00 00
TLS Address:                 00000035016dae00
TLS Dump:                                 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             0035016dae00 06 00 10 00 0c 00 00 80 01 00 00 00 cc 81 26 00
                             0035016dae10 00 00 00 00 50 02 00 00 30 b8 74 7f 10 00 00 50
                             0035016dae20 53 46 43 49 01 00 00 00 02 00 00 00 00 00 00 00
                             0035016dae30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0035016dae40 04 00 69 01 1e 00 10 c0 89 00 00 00 00 00 00 00
                             0035016dae50 10 00 00 00 70 aa 28 d4 04 00 10 00 00 00 00 00
                             0035016dae60 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0035016dae70 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0035016dae80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0035016dae90 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0035016daea0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0035016daeb0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0035016daec0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0035016daed0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0035016daee0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0035016daef0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

SciresM, 2022-01-08:
1) you did not use the issue templates

2) Remove custom sysmodules/etc.

closed 2022-01-08 by SciresM

Darkdragon20101, 2022-01-08:
It doesn't solve the issue and sorry i post the issue with the template