Atmosphere-NX/Atmosphere #1759: Official system updates managed by atmosphere

Diliz, 2022-01-10:
Hello!

I have a naive question regarding the official system updates, is there a possibility to manage official os updates from atmosphere?

This would fix 2 issues:
- No more update prompt anymore on game launch ( if possible, it's so annoying)
- No more failed updates on emunand.

If we could configure a default updater application and update with it when a new system update is available, it would fix the emunand corruption issues, even if the current atmosphere is not compatible with it yet, it would greatly simplify the update process afterward, without corrupting the emunand :)

At least having the possibility to remove the update prompt on game launch as a quick fix, this would solve the pebkac issues when peoples click on update instead of just launch the game.

Thanks for reading, and thanks for your work,
Diliz

Masamune3210, 2022-01-10:
We have daybreak but I dont think there is a way to redirect the flow to it or anything like that, at least yet. Not sure its even possible as settings is just part of home menu, its not a normal applet or app like everything else

SciresM, 2022-01-10:
This just kinda isn't how updates work on switch.

closed 2022-01-10 by SciresM

Diliz, 2022-01-10:
Thanks for the answer, as I said, it was a naive question :)
I tought the update prompt would at least be blockable

SciresM, 2022-01-10:
The update prompt is mostly out of our control tbh.

However, if you're getting the update prompt, it means you're connected to the internet...so why not just install the update directly from Nintendo? 

That's how you're intended to be updating with atmosphere (and then you can dump the installed update and use daybreak to install it on emummcs if you want).

Updating will cause issues for the first { days } after a major system update, but I do put a lot of effort into making sure that the currently-released atmosphere version always supports the latest firmware update.

Diliz, 2022-01-10:
I understand, my use case was mostly to stay on emunand if possible and at least not "misclick" on the update prompt when it appears, making it mistake proof for fast homebrew launch like retroarch, which is actually what a mate did last week xD

Anyway, thanks for your answers SciresM, and thanks for your hard work there again :)