Atmosphere-NX/Atmosphere #1762: A fatal error occurred when running atmospere 1.2.5

Lordloco23, 2022-01-13:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Atmosphère 1.2.5 
Fw 13.02

Error desc : kernel panic  0xf00
Fatal errors report :00000000017a8fab.bin

SciresM, 2022-01-13:
This is the third time you have made this issue after having one of the two duplicates you made closed for piracy/not filling out the issue template/not uploading the report file. I suspect you aren't actually reading what I'm saying.

closed 2022-01-13 by SciresM

Lordloco23, 2022-01-13:
Entschuldigen Sie, bin ganz neu hier, mein Englisch ist auch nicht das
beste, lebe in Deutschland. Dachte man könnte mir hier helfen, nachdem ich
gegoogelt habe

Masamune3210, 2022-01-13:
Sorry, I'm brand new here, my English isn't that either
 best, live in Germany.  Thought you could help me here after I did
 Googled

Verwenden Sie Google Translate, Sie müssen zumindest versuchen, andere zu verstehen, sonst wird Ihnen niemand helfen

Piraterie wird nicht unterstützt. Wenn Sie also Hilfe benötigen, müssen Sie Ihren Atmosphärenordner löschen und einen sauberen besorgen